socallinuxexpo/scale-network #497: Fix switch bulk loader defaults and other assumptions

sarcasticadmin, 2022-07-29:
## Description
Introduced in: https://github.com/socallinuxexpo/scale-network/pull/492

`owen` user set in bulker loader when it should be generic

## Acceptance Criteria
- Any assumptions introduced in https://github.com/socallinuxexpo/scale-network/pull/492 should be corrected

owendelong, 2022-07-30:
Need to find better solutions to several things in bulk loader...

User should probably be a command line parameter with default to blank (i.e. use .ssh/config value or local username).
(Requires SSH key be present in miniconfig (currently {admin, owen})

Currently throws commands at switch and assumes success unless ssh exits badly.
Should probably use expect.pm and care about results.

Way too chatty.

Should store up errors and produce summary at the end.

owendelong, 2023-09-13:
This is closely related to the work which needs to be done to override_config #499 .

owendelong, 2024-01-28:
Resolved via deprecation of bulk loader in favor of new loader library and script built on expect.pm.

closed 2024-01-28 by owendelong