socallinuxexpo/scale-network #508: Dedicated loghost for rsyslog log aggregation

sarcasticadmin, 2022-12-24:
## Description
Historically this was setup on `server3` (alias `bigbrother` or `monitoring` server). See DNS entries here: https://github.com/socallinuxexpo/scale-network/blob/19c9238e7482b35ca9bef3673919005fc21ff895/facts/servers/examples/dns/db.scale.lan.records.servers#L16-L18

It would be nice to have a dedicated machine for aggregating logs since the monitoring server still has to be figured out along with existing items like: https://github.com/socallinuxexpo/scale-network/issues/240 and the load requirements are already on the high side.

This loghost machine will also provide as good proving ground for getting together our framework for building and deploying with nixos instead of ansible.

## Acceptance Criteria
- Create a machine that provides a minimal rsyslog server to support existing log aggregation needs by scale-network

davidelang, 2022-12-24:
one of the reasons for useing the monitoring server for logging is that there 
are things that show up in the logs that we want to scrape and graph, or 
otherwise report

not that it can't be done on a separate VM, just explaining why it was done this 
way to begin with.

David Lang

sarcasticadmin, 2022-12-24:
Thanks @davidelang for the clarification, that makes sense based on the past conferences

owendelong, 2022-12-24:
I’m going to suggest an alternative here. What if we had a dedicated instance for persistent storage that was shared across all VMs. Dedicated log server would write logs to a section of that server for logs, but others could read it for scraping/graphing/etc.This would only be for things we want to store persistently across years/upgrades/etc. In this way, all of the other VMs could undergo “bare metal” rebuilds without worrying about preserving content/config. Config would either come from the git repo and content would be in persistent storage. Thoughts?

PR #509 (2022-12-24): [REVIEW] - init flake.nix and loghost machine build
## Description of PR

Fixes: #508

Adding initial flake.nix and consuming the `shell.nix` from `devShells.default` which we added back in #395 . Additionally adding a new NixOS VM for a dedicated `loghost` server along with tests to validate it works inside qemu. Still need to try out the VM in bhyve but this is a good checkpoint from me to get feedback.

## Previous Behavior
- Loghost was setup on `server3`

## New Behavior
- flake.nix and flake.lock add for consistent reference to `nixpkgs`
- Reference shell.nix from `devShells.default`
- ` [... cut]
closed 2022-12-24 by owendelong