socallinuxexpo/scale-network #527: Raspberry Pi Kiosk

norsehorse89, 2023-01-29:
## Description
- Followed guide at https://www.raspberrypi.com/tutorials/how-to-use-a-raspberry-pi-in-kiosk-mode/
- SSH key reused from previous Ubuntu IoT config, but can generate new key if needed

## ToDo
- Browser sometimes fails to load (permissions issue?); my_script.sh written by Sergey to monitor, assigned in crontab
- Scripts switched to open Firefox, can be changed back using guide
- Still need script to determine sign or registration Pi; can determine from lsusb or DHCP

## Acceptance Criteria
What is the expected outcome that resolves this issue?

norsehorse89, 2023-01-29:
Previous Ubuntu Core config (for reference, NOT implemented due to inability to Netboot):
- Reference guide: https://mir-server.io/docs/make-a-secure-ubuntu-web-kiosk
- Downloaded ubuntu core 64-bit image (ubuntu-core-20-arm64+raspi.img.zx)
- Decompressed image file (ubuntu-core-20-arm64+raspi.img)
- dd file to sd card (bs=32M)
- boot raspberry pi with flashed sd card, rpi boots to a prompt for an email linked to a ubuntu SSO account
- created [scale.techteam@gmail.com](mailto:scale.techteam@gmail.com) google account with apple keychain-generated password
- created ubuntu SSO account (username scaletech) and validated email
- generated ssh key (ssh-keygen -t ecdsa -C [scale.techteam@gmail.com](mailto:scale.techteam@gmail.com)), passphrase [redacted]
- imported .ssh/id_ecdsa.pub to ubuntu SSO account
- linked ubuntu SSO email to rpi, rpi prompts to ssh into it
- ssh into rpi, install snaps (snap install ubuntu-frame wpe-webkit-mir-kiosk)
- set kiosk url (snap set wpe-webkit-mir-kiosk url=https://www.socallinuxexpo.org/)

tweak42, 2023-01-30:
Discussion and input from  Owen and David roughly came to these requirements.

Boots to Full screen browser
    Disable break out browser and X (lockdown keys)
Using DPI scaling for readability at native screen resolution if possible 
Netboot image
      File overlay - Will this work with netbooting?
      Limit browser cache - to reduce ram usage since because of netbooting
      Connected mouse or dhcp parameter determines loaded URL
Disable unneeded services
      Disable wifi 
Enable ssh
Zabbix agent
Syslog config to output to log server

sarcasticadmin, 2023-03-23:
Relates to #537

davidelang, 2023-03-23:
a note on ubuntu IoT, Owen and I looked at that a year or so ago and it could 
nto be converted to a netboot, it installs everything, including the kernel, as 
snap items, and buries the device/partition information inside the snap

that was very disappointing to us, it seemed so easy to get it setup as a kiosk.

David Lang

  On Wed, 22 Mar 2023, Robert James Hernandez wrote:

> Date: Wed, 22 Mar 2023 22:31:25 -0700
> From: Robert James Hernandez ***@***.***>
> Reply-To: socallinuxexpo/scale-network
>     ***@***.***>
> To: socallinuxexpo/scale-network ***@***.***>
> Cc: Subscribed ***@***.***>
> Subject: Re: [socallinuxexpo/scale-network] Raspberry Pi Kiosk (Issue #527)
> 
> Relates to #537
>
>

sarcasticadmin, 2023-03-23:
@davidelang thats too bad

Good news is that @MatthewCroughan already has a pretty slick kiosk setup from the work that @MrHamel and him did over the weekend of 20x. I know hes polishing up the last little bits so we should see a PR raised for this soon.

norsehorse89, 2023-04-03:
Thinking about whether this issue should be moved to https://github.com/socallinuxexpo/scale-signs, as it contains the majority of Kyle's code (especially https://github.com/socallinuxexpo/scale-signs/tree/master/client/usr/bin)

davidelang, 2023-04-03:
The pi kiosk is used for the signs, but it's also used for registration, and in 
the future I hope to see it used for an alternate sign that shows off network 
monitoring stuff...

This is really a pretty generic browser kiosk, not tightly coupled with the 
existing sign software.

I could see a future version of sign software that had much more logic on the pi 
and less on the server side, but that would be a very significant re-write and 
re-architecture of the system.

David Lang

  On Sun, 2 Apr 2023, caleb wong wrote:

> Thinking about whether this issue should be moved to https://github.com/socallinuxexpo/scale-signs, as it contains the majority of Kyle's code (especially https://github.com/socallinuxexpo/scale-signs/tree/master/client/usr/bin)
>
>

nixinator, 2023-04-03:
Anything is possible with time , money and nix.

however, only time and money will take you so far :-)

owendelong, 2023-09-13:
It seems Ubuntu is going deep down the SNAP rabbit hole even in their desktop versions, so I think it is time to abandon the Ubuntu rabbit hole.

davidelang, 2023-09-13:
I thought we were using raspian (raspberry OS, whatever they are calling it 
now), which is debian, not ubuntu.

and then I thought the nixos folks were replacing everything

David Lang

On Tue, 12 Sep 2023, Owen DeLong wrote:

> It seems Ubuntu is going deep down the SNAP rabbit hole even in their desktop versions, so I think it is time to abandon the Ubuntu rabbit hole.

owendelong, 2023-09-13:
We at least were using Ubuntu for some desktop things and other purposes in the NOC.

I’d like to make sure we aren’t doing that any more as they go deep down the SNAP rabbit hole.

Owen

MatthewCroughan, 2023-09-13:
Happy to help with this, I'll speak to @sarcasticadmin about what to do.

MatthewCroughan, 2023-09-13:
Yeah this was all done last time here https://github.com/socallinuxexpo, the todo explains what's left to do and how to reproduce what's there. It's just the PXE booting left to do, which is something I can work on at Nix Camp with @sarcasticadmin

MatthewCroughan, 2023-09-13:
FYI the current implementation I just posted uses `cage` which is a wayland compositor that makes escaping to the TTY impossible, as far as I can tell, so that is handled, since that is the primary function of `cage`. https://github.com/socallinuxexpo/scale-kiosk/blob/master/kiosk.nix

sarcasticadmin, 2023-09-13:
> Yeah this was all done last time here https://github.com/socallinuxexpo/scale-kiosk, the todo explains what's left to do and how to reproduce what's there. It's just the PXE booting left to do, which is something I can work on at [Nix.Camp](https://nix.camp) with @sarcasticadmin

Happy to work on this in the coming weeks while were together at nix camp. I know were pretty much there for creating an image from all the work that @MatthewCroughan and @MrHamel worked on during 20x

sarcasticadmin, 2024-03-02:
Adding some notes here since I found the following requirements out after talking with Lei:

1. Modified the user-agent - Currently, the user agent has to contain "SCALE:NNN", where NNN is the last octet in the IPv4 address. You can see an example of this in the old `reg.py` script: https://github.com/socallinuxexpo/scale-network/blob/d695fc5f1121b4715a0b602508e4d0fe600a8d39/pi/scripts/reg.py#L32 and setting it here: https://github.com/socallinuxexpo/scale-network/blob/d695fc5f1121b4715a0b602508e4d0fe600a8d39/pi/scripts/reg.py#L9-L19

3.  set browser `dom.allow_scripts_to_close_windows` or equivalent - Here the workflow is a new attendee comes to use the kiosk, when they are done they should be able to click the close window button in the upper right corner which causes the window to close, then our script should just relaunch another instance of the browser with the default registration page. This ensures that any data entered is ephemeral 

The pi4 kiosk can be in the following states and all need to be tested on the hardware itself:

```
1. network && keyboard
2. no network && keyboard
4. network && no keyboard
5. no network && no keyboard
```

We were able to satisfy and confirm all of these tests via https://github.com/socallinuxexpo/scale-kiosk/pull/2

We still have some small changes to make, but the majority functionality for the 2 Kiosk requirements is now in place

sarcasticadmin, 2024-03-25:
This was completed for 21x and the results of the work can be found: https://github.com/socallinuxexpo/scale-kiosk

closed 2024-03-25 by sarcasticadmin