socallinuxexpo/scale-network #532: Belkin RT3200 OpenWrt conversion

ptrlow1, 2023-01-30:
## Description
As we intend to use the RT3200s to run OpenWrt from now on, we need to convert these to "UBI" layout before flashing the image.

https://openwrt.org/toh/linksys/e8450#installation

## Acceptance Criteria
Convert to UBI layout by using this source from this git repo: 
https://github.com/dangowrt/linksys-e8450-openwrt-installer

ptrlow1, 2023-03-09:
Note that this is an inital flash. (We **only** do this **once**.) After that, it will be part of the auto AP mass flash operation we do at work parties.

ptrlow1, 2023-03-10:
Finished converting the 3200s to OpenWrt late Wednesday night (right before they threw us out). Rob will finish those with our built image, and will be deployed to the show.

ptrlow1, 2023-03-23:
Images built by Rob didn't work as they were built without "UBI" support. 

https://openwrt.org/docs/techref/flash.layout#ubi_unsorted_block_images

sarcasticadmin, 2023-08-03:
Posting some stock configs for reference

/etc/config/wireless:
```

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/18000000.wmac'
	option channel '1'
	option band '2g'
	option htmode 'HT20'
	option disabled '1'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option channel '36'
	option band '5g'
	option htmode 'HE80'
	option disabled '1'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'
```

/etc/config/network:

```

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fda4:b995:95b4::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option device 'wan'
	option proto 'dhcp'

config interface 'wan6'
	option device 'wan'
	option proto 'dhcpv6'
```

/etc/config/system:

```
config system
	option hostname 'OpenWrt'
	option timezone 'UTC'
	option ttylogin '0'
	option log_size '64'
	option urandom_seed '0'

config timeserver 'ntp'
	option enabled '1'
	option enable_server '0'
	list server '0.openwrt.pool.ntp.org'
	list server '1.openwrt.pool.ntp.org'
	list server '2.openwrt.pool.ntp.org'
	list server '3.openwrt.pool.ntp.org'

config led 'led_wan'
	option name 'WAN'
	option sysfs 'inet:blue'
	option trigger 'netdev'
	option mode 'link tx rx'
	option dev 'wan'

```

/etc/os-release:

```
NAME="OpenWrt"
VERSION="22.03.1"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 22.03.1"
VERSION_ID="22.03.1"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r19777-2853b6d652"
OPENWRT_BOARD="mediatek/mt7622"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 22.03.1 r19777-2853b6d652"
```

ptrlow1, 2023-08-18:
Note that the RT3200 only has a Power and Internet LED, there's no ethernet, wifi, or WPS LEDs. 

[https://github.com/socallinuxexpo/scale-network/blob/master/openwrt/docs/BUILD.md#build-identity](https://github.com/socallinuxexpo/scale-network/blob/master/openwrt/docs/BUILD.md#build-identity)
> Depending on the SCaLE conference number, the WPS LED will be ON for even years and OFF for odd years.
> 
> NOTE: This depends on the SCaLE conference number from facts.yaml not the year since the build would drift based on when it was built.

owendelong, 2023-08-19:
True, but they are both triple LEDs capable of rendering multiple colors, therefore we could use different colors of internet LED (or even power LED) potentially for conference version signaling.

I’ll note that on the RT3200, the factory firmware uses the power LED to indicate WPS pairing enabled.

Owen

sarcasticadmin, 2023-11-08:
Closing as this was completed via #611

closed 2023-11-08 by sarcasticadmin