socallinuxexpo/scale-network #537: RPi PXE Boot

MrHamel, 2023-02-18:
## Description
Add RPi PXE Booting to Automation

## Acceptance Criteria
What is the expected outcome that resolves this issue?

Be able to build a VM with the necessary config files and packages to PXE boot a number of RPi's.

GitHub is picky about file extensions used when uploading files to issues.

[exports.txt](https://github.com/socallinuxexpo/scale-network/files/10775065/exports.txt) -> /etc/exports
[kea-dhcp4.conf.txt](https://github.com/socallinuxexpo/scale-network/files/10775064/kea-dhcp4.conf.txt) -> /etc/kea/kea-dhcp4.conf

It took me awhile to get to this due to physical therapy, lost my job, and prior arrangements. I have a complete image of "server5" so I can grab anything that is needed or share the partition image.

While making the image, I've learned the hard way that Clonezilla's use of `partclone` does not produce a loop mountable partition images by default even if it's RAW type. Luckily `partclone.restore` was able to remake the file into something that can be mounted with ease.

I'm currently in the process of `xz` compressing the sda3 raw partition file as I write this, and will upload it to my server at the colo for anyone who wishes to download it.

davidelang, 2023-03-08:
following up on this, are the individual server images posted anywhere? we need 
them right away if we are going to use them this year.

David Lang

On Sat, 18 Feb 2023, Ryan Hamel wrote:

> Date: Sat, 18 Feb 2023 11:56:31 -0800
> From: Ryan Hamel ***@***.***>
> Reply-To: socallinuxexpo/scale-network
>     ***@***.***>
> To: socallinuxexpo/scale-network ***@***.***>
> Cc: Subscribed ***@***.***>
> Subject: [socallinuxexpo/scale-network] RPi PXE Boot (Issue #537)
> 
> ## Description
> Add RPi PXE Booting to Automation
>
> ## Acceptance Criteria
> What is the expected outcome that resolves this issue?
>
> Be able to build a VM with the necessary config files and packages to PXE boot a number of RPi's.
>
> GitHub is picky about file extensions used when uploading files to issues.
>
> [exports.txt](https://github.com/socallinuxexpo/scale-network/files/10775065/exports.txt) -> /etc/exports
> [kea-dhcp4.conf.txt](https://github.com/socallinuxexpo/scale-network/files/10775064/kea-dhcp4.conf.txt) -> /etc/kea/kea-dhcp4.conf
>
> It took me awhile to get to this due to physical therapy, lost my job, and prior arrangements. I have a complete image of "server5" so I can grab anything that is needed or share the partition image.
>
> While making the image, I've learned the hard way that Clonezilla's use of `partclone` does not produce a loop mountable partition images by default even if it's RAW type. Luckily `partclone.restore` was able to remake the file into something that can be mounted with ease.
>
> I'm currently in the process of `xz` compressing the sda3 raw partition file as I write this, and will upload it to my server at the colo for anyone who wishes to download it.
>
>

MrHamel, 2023-03-08:
Server images meaning the RPi? They are files. I was lead to believe that another NFS rootfs was in the works for the RPi's?

If you're referring to sda3 partition of "server5", I dropped the ball on getting the image uploaded, it'll be available at https://ryanhamel.com/scale/sda3-ext4.raw.xz in 90 minutes (when the SCP will complete). File is ~22GB in size, going over a 40Mbit cable upload.

I can bring the server over to the convention center tomorrow afternoon, would be best to run off that for this year, and pick back up on virtualizing it after the conference. I have been making job hunting a much higher priority.

davidelang, 2023-03-08:
the new pi image did not get created, so we want to use the one from that 
server.

we could recreate that pi image withut too much hassle, but the initrd to boot 
from is more work.

David Lang

On Tue, 7 Mar 2023, Ryan Hamel wrote:

> Date: Tue, 07 Mar 2023 22:53:10 -0800
> From: Ryan Hamel ***@***.***>
> Reply-To: socallinuxexpo/scale-network
>     ***@***.***>
> To: socallinuxexpo/scale-network ***@***.***>
> Cc: David Lang ***@***.***>, Comment ***@***.***>
> Subject: Re: [socallinuxexpo/scale-network] RPi PXE Boot (Issue #537)
> 
> Server images meaning the RPi? They are files. I was lead to believe that another NFS rootfs was in the works for the RPi's?
>
> If you're referring to sda3 partition of "server5", I dropped the ball on getting the image uploaded, it'll be available at https://ryanhamel.com/scale/sda3-ext4.raw.xz in 90 minutes (when the SCP will complete). File is ~22GB in size, going over a 40Mbit cable upload.
>
> I can bring the server over to the convention center tomorrow afternoon, would be best to run off that for this year, and pick back up on virtualizing it after the conference. I have been making job hunting a much higher priority.
>
>

MrHamel, 2023-03-08:
@davidelang Fixed and tested, was a permissions issue.

MrHamel, 2023-03-08:
The other partitions to go along with it:

https://ryanhamel.com/scale/sda1.dd-ptcl-img.uncomp.xz
https://ryanhamel.com/scale/sda2-vfat.raw.xz

Below are the partition tables in various formats, really starting to regret not doing just a simple `dd`.

```
# cat sda-pt.parted
Model: ATA WDC WD2500AAKX-0 (scsi)
Disk /dev/sda: 488397168s
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags: 

Number  Start     End         Size        File system  Name                  Flags
 1      2048s     4095s       2048s                                          bios_grub
 2      4096s     1054719s    1050624s    fat32        EFI System Partition  boot, esp
 3      1054720s  488396799s  487342080s  ext4

# cat sda-gpt.sgdisk
Disk /dev/sda: 488397168 sectors, 232.9 GiB
Model: WDC WD2500AAKX-0
Sector size (logical/physical): 512/512 bytes
Disk identifier (GUID): 85228006-556E-4166-907C-A6651876FA1A
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 488397134
Partitions will be aligned on 2048-sector boundaries
Total free space is 2349 sectors (1.1 MiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048            4095   1024.0 KiB  EF02  
   2            4096         1054719   513.0 MiB   EF00  EFI System Partition
   3         1054720       488396799   232.4 GiB   8300
 
# cat sda-pt.sf       
label: gpt
label-id: 85228006-556E-4166-907C-A6651876FA1A
device: /dev/sda
unit: sectors
first-lba: 34
last-lba: 488397134
sector-size: 512

/dev/sda1 : start=        2048, size=        2048, type=21686148-6449-6E6F-744E-656564454649, uuid=FD2209DA-1D36-43E7-8865-FFBD8FA78395
/dev/sda2 : start=        4096, size=     1050624, type=C12A7328-F81F-11D2-BA4B-00A0C93EC93B, uuid=ADB56AFA-5CB7-4A34-892A-0678EE1D9AB0, name="EFI System Partition"
/dev/sda3 : start=     1054720, size=   487342080, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=A057B41C-4A9B-43A2-ADBC-DBE5FE613159
```

MrHamel, 2023-03-08:
I am available to bring the server over to the convention center this afternoon, or hand it to someone before/after dinner.

owendelong, 2023-09-13:
Has there been any further progress on this? Is there a way we can make progress before the show?

owendelong, 2024-12-16:
It looks like some progress may have been made:
https://forums.raspberrypi.com/viewtopic.php?t=378247

owendelong, 2024-12-16:
And
https://www.raspberrypi.com/documentation/computers/remote-access.html#network-boot-using-ipv6

sarcasticadmin, 2025-01-27:
@owendelong did some more debugging and opened up this in the rpi firmware repo: https://github.com/raspberrypi/firmware/issues/1938