socallinuxexpo/scale-network #560: Consider disconnecting wifi clients with weak signals

genebean, 2023-03-10:
## Description

In general, many wireless clients (especially phones) do not like to let go of an access point even when a stronger signal is available. In this case, it can be helpful to kick off clients with too weak of a signal so as to force them to connect again and choose an AP with a higher signal strength.

## Suggestion

Much search lead me to https://github.com/barbieri/barbieri-playground/tree/master/openwrt/wifi-disconnect-low-signal. Initial reading of this sounds like it would be worth a try.

nixinator, 2023-11-23:
I think the AP's have a bit more horse power, so this is definitive option... This make roaming a little more reliable.

However, this will generate more disassociations, and put some more load on our DHCP servers as wireless clients move around.

owendelong, 2023-11-26:
Load on the DHCP server will be minimal. The users aren’t changing VLAN when they move, so worst case should be just a renew. I don’t see that as an issue worthy of consideration.

nixinator, 2023-11-27:
Hmmm, okay, i was under the impression when wifi clients dissociate and associate they will renew their lease on wifi connect.  

This would be a nice feature to implement.

But if the network can handle it, then lets do it.

owendelong, 2023-11-27:
Yes, but a renew (vs. a discover) is a pretty low-overhead operation — Client sends renew request, server checks current lease database for validity, sends back new lease, updates lease database, done.

Vs. a discover which is several more packets exchanged and quite a bit more overhead and sanity checking on the server side.

Owen