socallinuxexpo/scale-network #564: rename expo floor switches/aps

davidelang, 2023-03-10:
current names are 
A wall
then A - C (front to back) and 1-5 (typically left to right)

problems:
booths are numbered right to left
 sort order sorts by front to back, not side to side, when there are only 4 rows + wall  the 'missing ones' show up in the middle of the list.

solution:

next year have the letter be the left/right direction and the number be the front-back position.

owendelong, 2023-03-26:
The problem with this theory is that we define 3 letters (front/middle/back) and 5 numbers + wall (to support up to 5 rows).

How about the solution is we just place the numbers (1/2/3) in the corresponding order with the booth aisle numbers.

davidelang, 2023-03-26:
On Sat, 25 Mar 2023, Owen DeLong wrote:

> The problem with this theory is that we define 3 letters (front/middle/back) and 5 numbers + wall (to support up to 5 rows).
>
> How about the solution is we just place the numbers (1/2/3) in the corresponding order with the booth aisle numbers.

the problem is that the number of rows (digits) varies. This year it was 4 + 
wall, a few years ago it was 5+ wall. With the highest number being against the 
wall. given that we are doing 1 on the wall, 2 on the row next to it, 1 in the 
next row, etc, numbering right to left is a problem

for the record, this is assuming that the show ends up with a similar layout in 
the future, we've seen it change before.

David Lang

owendelong, 2023-03-26:
It really isn’t. This year, for example, we had 4 rows each with three switches and one on the watch. This was [1234][ABC] plus WallA. 

We placed the 1-4 switches in the opposite row order, but there was no need to do so. The switches for non-existent row 5 didn’t get deployed and weren’t needed. 

Owen

davidelang, 2023-03-26:
so how do you number the APs so that they work with both 4 and 5 rows?

David Lang

On Sun, 26 Mar 2023, Owen DeLong wrote:

> It really isn’t. This year, for example, we had 4 rows each with three switches and one on the watch. This was [1234][ABC] plus WallA.
>
> We placed the 1-4 switches in the opposite row order, but [...]

owendelong, 2023-09-13:
IMHO, number them 1 through 5 and simply don't deploy row 5 if it's not needed. Is there some problem with this deployment that I'm missing?

owendelong, 2023-09-13:
The real issue was deploying the switches in the incorrect order (Row 1 switches got deployed in row 4 and vice versa, siimilarly 2 and 3 were swapped). If we deployed in the order Wall...1...2...3...4...[5]...[6]...??? I think we'd be fine (which is what was intended, but not what happened).

owendelong, 2023-09-13:
I'm ready to close this as a training issue rather than a numbering/naming problem unless you have further comment, David.

davidelang, 2023-09-13:
On Tue, 12 Sep 2023, Owen DeLong wrote:

> The real issue was deploying the switches in the incorrect order (Row 1 switches got deployed in row 4 and vice versa, siimilarly 2 and 3 were swapped). If we deployed in the order Wall...1...2...3...4...[5]...[6]...??? I think we'd be fine (which is what was intended, but not what happened).

that's just not how they are planned currently (and it's not symmetrical to just 
allow us to flip it), which is why I created this ticket so that we could change 
the plan to something like you described which would work.

David Lang

owendelong, 2023-09-13:
What I described WAS the plan all along.

Even if not symmetrical, it was still SUPPOSED to be Wall…1…2…3…4… etc.

With A near the street and C near the Theater.

Owen

closed 2024-01-28 by owendelong