socallinuxexpo/scale-network #567: monitoring and make nice cute graphs of network ....stuff

nixinator, 2023-03-10:
## Description

Monitoring of various network stats need to be configured.

## Acceptance Criteria

Nice graphs!!!

nixinator, 2023-03-24:
so doing some more research into this it seems

collectd running on the openWRT AP's feeding back to influx and graphana might be a good option.

It's quite declarative , and the configuration can be built from the single source from the scale-network repo.

https://blog.christophersmart.com/2019/09/09/monitoring-openwrt-with-collectd-influxdb-and-grafana/

We can probably monitor a whole  host of other things using this method.

I'll have a play and see what it looks like.

sarcasticadmin, 2023-03-24:
> collectd running on the openWRT AP's feeding back to influx and graphana might be a good option.
>
> It's quite declarative , and the configuration can be built from the single source from the scale-network repo.

I saw this in the past and it looks like a promising solution, especially if the entire config is declarative :+1: 

> I'll have a play and see what it looks like.

Let me know how it goes :+1:

sarcasticadmin, 2024-01-25:
Given that were utilizing prometheus a good dashboard starting place might be: https://grafana.com/grafana/dashboards/11147-openwrt/

sarcasticadmin, 2024-03-25:
Closing this as the initial work for getting grafana up and running has been completed. Any discuss around specific graphs should be discussed in #756

closed 2024-03-25 by sarcasticadmin