socallinuxexpo/scale-network #583: simplifying the static ip address for nixos machines

sarcasticadmin, 2023-03-11:
## Description

Currently we are hardcoding the static ip address in each machines configuration. There should be a more uniform way of address assignment so that we can pass around these facts to all the machines that are defined in the flake.

## Acceptance Criteria
- static addresses arent hardcoded in each machine specific configuration

davidelang, 2023-03-11:
I'm perfectly happy with only the dhcp server being truely static, with the 
others getting addresses via dhcp reservations (as long as we can tweak the boot 
process so that they startup a few seconds after the dhcp server)

David Lang

owendelong, 2023-04-02:
Actually, if we set up their boots to wait indefinitely for DHCP, that should be adequate and not require staged or ordered booting of the VMs.

owendelong, 2024-02-11:
Also, the VM Host (hypervisor) should have static addressing as well.

owendelong, 2024-03-01:
Rob and I discussed this today. It probably won't happen this time around, but the long term plan is to static address (hand coded) the Hypervisors and the DHCP servers and go with DHCP/DHCP6 reservations for everything else.

The DHCP servers should be the first VMs booted on the hypervisor, but they should boot fast enough that the other machines will still be well within their retry window by the time they start looking for addresses.

owendelong, 2024-12-09:
Now that we are moving to mostly straight utilities on bare metal, I think one static address on the physical hosts (which will run the KEA instances) should be all that needs to be static. The rest can be handled through DHCP reservations. Since the two servers form an HA pair for all DHCP services, we shouldn't have an instance where the DHCP service is completely down unless we have bigger network problems that would be more impactful than a lack of addressing on hosts.

sarcasticadmin, 2024-12-09:
> Now that we are moving to mostly straight utilities on bare metal, I think one static address on the physical hosts (which will run the KEA instances) should be all that needs to be static.

That's correct, this is simplified dramatically from what we were running for 21x

> If we are willing to run NetworkMangler^wNetworkManager

Im voting `no` here. Its good to know, but seems like overkill

owendelong, 2024-12-16:
OK, sounds like we have a solution. Any reason not to close this issue?

kylerisse, 2025-02-25:
> OK, sounds like we have a solution. Any reason not to close this issue?

Agree with @owendelong . No reason I can think of to keep this open.

sarcasticadmin, 2025-02-25:
Ya we got rid of the microvms (https://github.com/socallinuxexpo/scale-network/pull/841) and are going to run on bare metal

closed 2025-02-25 by sarcasticadmin