socallinuxexpo/scale-network #589: CI job to build scale-signs docker image

kylerisse, 2023-03-13:
## Description
Right now the `scale-signs` docker image is built locally and uploaded by @sarcasticadmin to his personal docker hub

## Acceptance Criteria
- scale-signs docker image is built by Github Actions and automatically uploaded to ${SOME_REGISTRY}

relates to 
- #418
- #588

nixinator, 2023-03-23:
if this needs an older version of PHP in nix, then this 

https://github.com/fossar/nix-phps

and

https://discourse.nixos.org/t/local-dev-stack-for-php/26644

maybe helpful to you.

nixpkgs repo tends to race ahead with versions of things, and be super new, if this is a good or a bad thing remains to be seen.

is the latest software the best software?

kylerisse, 2023-03-24:
hey that's a cool repo for pinning PHP, thanks @nixinator !

it looks like it only goes back to 5.6 thought. That might work. It currently runs on 5.4 in docker. 

I'll try 5.6 first and if that doesn't work I was thinking of just doing something like the below in the pkg to pin to an older revision of nixpkgs that still had 5.4. Then use it in the derivation with a modern version of apache. What do you think?

```
let
     oldpkgs = import (builtins.fetchGit {            
         name = "nixpkgs-from-20160120";                                                 
         url = "https://github.com/NixOS/nixpkgs/";                       
         ref = "refs/heads/nixpkgs-unstable";                     
         rev = "2f6ef9aa6a7eecea9ff7e185ca40855f36597327";                                           
     }) {};                                                                           

     myPkg = oldpkgs.php54;
in
```

nixinator, 2023-03-24:
wow! it's so old that it not even support in the old repo.

yeah, that could work. Nix is very good at reproducing old pieces of software.

So it might just work.

However, i think we all know the long term solution, is to upgrade these scripts to a modern version of PHP.

That's probably quite a lot of work, but it's going to need to be done at some point.

if the upgrade is going to be tough and lots of things needs to be upgraded... , maybe it's time to think about a new signs server, but that may be more of a socio-economic one that a technical issues.

davidelang, 2023-03-24:
The entire signs server needs to be re-written. It has not been maintained for a 
long time, and now that we have 4k screens everywhere we should take advantage 
of them (as just one minor thing)

On Thu, 23 Mar 2023, Lee Hughes wrote:

> wow! it's so old that it not even support in the old repo.
>
> yeah, that could work. Nix is very good at reproducing old pieces of software.
>
> So it might just work.
>
> However, i think we all know the [...]

kylerisse, 2023-03-24:
Agreed @davidelang but until that happens let’s assume we need to keep running the existing one.

sarcasticadmin, 2023-04-15:
Another note before I forget: Lets make sure we move the credentials we lay down to somewhere that's persistent instead of `/var` where I currently have them: https://github.com/socallinuxexpo/scale-network/blob/d890d073a2d3c8110ba47844cca85f5106e39bea/nix/machines/signs.nix#L20

kylerisse, 2023-04-15:
> Another note before I forget: Lets make sure we move the credentials we lay down to somewhere that's persistent instead of `/var` where I currently have them:
> 
> https://github.com/socallinuxexpo/scale-network/blob/d890d073a2d3c8110ba47844cca85f5106e39bea/nix/machines/signs.nix#L20

You captured this in #588 but with a different path name

sarcasticadmin, 2023-04-15:
> You captured this in https://github.com/socallinuxexpo/scale-network/issues/588

I guess i did :laughing: Ill leave #588 open for now, but if/when we pick this one up the reference will be good

kylerisse, 2025-09-30:
Discussed with @sarcasticadmin at the last Work Party. We're not going to touch this for `23x`. The goal this year is to run [go-signs](https://github.com/kylerisse/go-signs) on the pis. We will also run, as a backup, [scale-signs](https://github.com/socallinuxexpo/scale-signs) on the server using the current container config. Based on how this year goes, my preferred outcome is that we retire `scale-signs` and move on to `go-signs`, which is already Nix native. Here is a [sample implementation](https://github.com/kylerisse/nixcfg/blob/master/machines/gibson/configuration.nix#L53-L57). At that point, we can close this issue. Please add any feedback or concerns with this plan.