socallinuxexpo/scale-network #593: enable ntp dhcp option in kea

sarcasticadmin, 2023-03-15:
## Description

Blocked by: #592 

Now that we have a ntp service lets define the correct dhcp option in kea so ensure that clients are getting this information.

## Acceptance Criteria
- `kea` serving up ntp servers via dhcp options
- tests to confirm that `kea` is correctly setting those options for a client

owendelong, 2024-01-28:
Does this still need something? How is our Kea configuration built and where would the options need to be added? Let's do this for both v4 and v6 at the same time.

sarcasticadmin, 2024-02-19:
Adding some notes here about dhcpv6:

There seem to be more than one way to set NTP servers via DHCPv6: https://superuser.com/a/1766773 which calls out RFC 5908: https://www.rfc-editor.org/rfc/rfc5908.html#section-7

Kea only seems to be aware of SNTP (option 31): https://kea.readthedocs.io/en/kea-2.0.0/arm/dhcp6-srv.html#id2

We could also set both dhcp ntp options just to cover our basis: 31 & 56

MrHamel, 2024-06-01:
Hijacking this due to the RPi systemd-timesyncd delay issues when booting NixOS. Let's do that.

It seems very simple to add: https://github.com/socallinuxexpo/scale-network/blob/c6e657d6cca7fc17aac370905dd1e8105d087f8b/facts/inventory.py#L528

Signs is a v6 only network: https://github.com/socallinuxexpo/scale-network/blob/c6e657d6cca7fc17aac370905dd1e8105d087f8b/switch-configuration/config/vlans.d/Conference#L9 + https://github.com/socallinuxexpo/scale-network/blob/c6e657d6cca7fc17aac370905dd1e8105d087f8b/switch-configuration/config/vlans.d/Expo#L9

The Pi solution would be to either hard code the NTP server IPs into their config, update the Kea config with the applicable v6 core IPs + get the RPi config to ignore the built-in list and use DHCP https://nixos.wiki/wiki/NTP.

MrHamel, 2024-06-01:
https://wiki.archlinux.org/title/systemd-timesyncd#Configuration

Per the documentation, this is the order for NTP server list:

> Further to the daemon configuration, NTP servers may also be provided via a [systemd-networkd](https://wiki.archlinux.org/title/Systemd-networkd#[Network]) configuration with a NTP= option or, dynamically, via a DHCP server.
> 
> The NTP server to be used will be determined using the following rules:
> 
> - Any per-interface NTP servers obtained from [systemd-networkd.service(8)](https://man.archlinux.org/man/systemd-networkd.service.8) configuration or via DHCP take precedence.
> - The NTP servers defined in /etc/systemd/timesyncd.conf will be appended to the per-interface list at runtime and the daemon will contact the servers in turn until one is found that responds.
> - If no NTP server information is acquired after completing those steps, the NTP server host names or IP addresses defined in FallbackNTP= will be used.

Errata: According to the big cheese, it checks one server at a time (i.e no parallelism), so it's crucial that the first server be up for a quick RPi boot. -- https://github.com/systemd/systemd/issues/19494#issuecomment-831953675

An option around that if performance is important, is to ditch `systemd-timesyncd` for `chrony`, which would be forced to use hardcoded servers, and for backup it has a `pool` config option which allows it to discover and automatically maintain 2-16 servers from pool.ntp.org (more information at https://chrony-project.org/doc/4.1/chrony.conf.html).

If DHCP is a must for the RPi's we could pull off something like https://github.com/systemd/systemd/issues/19230#issuecomment-1972125246. The "if" is important, because I want to be sure `chrony` will attempt to sync in parallel, and whatever responds, wins that initial sync.

owendelong, 2024-06-01:
> Further to the daemon configuration, NTP servers may also be provided via a systemd-networkd <https://wiki.archlinux.org/title/Systemd-networkd#%5BNetwork%5D> configuration with a NTP= option or, dynamically, via a DHCP server.
> Errata: According to the big cheese, it checks one server at a time (i.e no parallelism), so it's crucial that the first server be up for a quick RPi boot. -- systemd/systemd#19494 (comment) <https://github.com/systemd/systemd/issues/19494#issuecomment-831953675>
> If DHCP is a must for the RPi's we could pull off something like systemd/systemd#19230 (comment) <https://github.com/systemd/systemd/issues/19230#issuecomment-1972125246>. The "if" is important, because I want to be sure chrony will attempt to sync in parallel, and whatever responds, wins that initial sync.

I don’t see any issue with adding it to the DHCP server.

Owen

owendelong, 2024-12-16:
Where do we stand on this? Did we ever get the namespace issue for kea on nix resolved so that we can either have them both use the same /var/lib and /var/run directories or at least make both sets accessible?

sarcasticadmin, 2025-09-10:
> Did we ever get the namespace issue for kea on nix resolved so that we can either have them both use the same /var/lib and /var/run directories or at least make both sets accessible?

Was fixed some time ago upstream in nixpkgs

> Where do we stand on this?

Currently dhcpv4 has this option already set in our config: https://github.com/socallinuxexpo/scale-network/pull/670

Last thing would be to set dhcpv6 `ntp-servers (option 56)`: https://kea.readthedocs.io/en/kea-3.0.0/arm/dhcp6-srv.html#id3

closed 2025-09-10 by sarcasticadmin