socallinuxexpo/scale-network #594: Possible AP Alternative: TP-Link EAP615-Wall

MrHamel, 2023-03-15:
Amazon: https://amazon.com/dp/B09KYK1S9R
OpenWRT: https://openwrt.org/toh/tp-link/eap615-wall
TP-Link: https://www.tp-link.com/us/business-networking/omada-sdn-access-point/eap615-wall/

A 2x2 MIMO 802.11.ax AP, very easy to install OpenWRT according to the provided article, has easy to use switch integration, and PoE passthrough on port eth3/lan3. The ports are all integrated into the Mediatek MT7621 ARM CPU, don't know if they will have any sort of hardware acceleration, would need to benchmark the throughput it to get an idea of expectations.

If deemed worthy for show use, special stands can be made up for them, which can also double as a small stand showing the password.

With the help of upgraded PoE switches in the IDF's, PoE passthrough can be useful for powering...

- A/V's camera, with the AP at the back of the room
- A sign RPi without an additional power strip or 3-way power adapter
- Daisy chaining up to 3 APs

Based on the room they're configured for, eth3/lan3 can be setup for the appropriate untagged (A/V or sign) or tagged (trunk to downstream APs) VLAN(s), eth1 as a trunk port for downstream switches, and eth2 a free port (probably best configured for untagged for guest Internet -- same VLAN as public Wi-Fi for security reasons). eth0 is of course the uplink port.

I am throwing this idea at the wall, seeing if it sticks. I know some team members have opinions on brands.

davidelang, 2023-03-15:
I've ordered one to experiment with.

any information on de-bricking these? one very nice thing about the wndr3800s 
that we've been using is that we can flash them no matter what software is on 
them.

David Lang

On Wed, 15 Mar 2023, Ryan Hamel wrote:

> Amazon: https://amazon.com/dp/B09KYK1S9R
> OpenWRT: https://openwrt.org/toh/tp-link/eap615-wall
> TP-Link: https://www.tp-link.com/us/business-networking/omada-sdn-access-point/eap615-wall/
>
> A 2x2 MIMO [...]

MrHamel, 2023-03-17:
Unfortunately I do not have that information. They run LEDE out of the box, which is why the only hurdle is disabling the firmware verification via SSH before flashing raw OpenWRT on them.

davidelang, 2023-03-17:
I ordered one and received it today, these are small, slightly thicker but 
otherwise smaller than my cell phone.

David Lang

On Thu, 16 Mar 2023, Ryan Hamel wrote:

> Unfortunately I do not have that information. They run LEDE out of the box, which is why the only hurdle is disabling the firmware verification via SSH before flashing raw OpenWRT on them.
>
>

MrHamel, 2023-03-18:
Between https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478 and https://github.com/openwrt/openwrt/pull/10238, it may be a good idea to upgrade to a snapshot which will have this patch merged in, helping out in benchmarks.

The 500Mbit ceiling is concerning when many devices on the Wi-Fi can bring down a switch or A/V, which this may resolve.

davidelang, 2023-03-18:
I don't think we have 500Mb uplink for the entire show to share, let alone a 
single AP. The cpu on this is a bit weak, not a lot more powerful than the 
wndr3800s we're using now. We make it work now by disabling conntrack and by 
doing simple bridging from wifi to vlan and not doing any additional packet 
processing on the AP.

David Lang

On Fri, 17 Mar 2023, Ryan Hamel wrote:

> Between https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478 and https://github.com/openwrt/openwrt/pull/10238, it may be a good idea to upgrade to a snapshot which [...]

davidelang, 2023-03-18:
ahh, this is a DSA issue.

Current versions of DSA only use a single ethernet connection between the cpu 
and the switch, even if the hardware has two connections wired. This is part of 
the 'hide the switch from the user' mentality of DSA (and the thing that Lee 
said about openwrt adding a tag when things arrive is a DSA thing.

The first version of DSA ( 
https://legacy.netdevconf.info/2.1/papers/distributed-switch-architecture.pdf ) 
effectively used the switch as just a port multiplexer, but later updates have 
enabled 'acceleration', which allows routing through the switch without touching 
the cpu, closer to what I did in the past with the swconfig approach.

This patch is a hack to enable one specific use case (it ties the 2nd cpu port 
to the 'wan' port on the switch), I don't expect it to go much of anywhere in 
the face of DSA (although it may encourage people to support using the 2nd 
interface)

In our case, the main bandwidth usage is wifi <-> wired, so if we are only using 
one ethernet, we are cpu limited and the data only passes the switch interface 
once, so it won't have this problem. (the wifi <-> cpu interface is separate 
than the cpu <-> switch interface, so the cpu issues I mentioned earlier are 
still concerns)

David Lang

On Fri, 17 Mar 2023, David Lang wrote:

> I don't think we have 500Mb uplink for the entire show to share, let alone a
> single AP. The cpu on this is a bit weak, not a lot more powerful than the
> wndr3800s we're using now. We make it work [...]

ptrlow1, 2023-03-23:
I mentioned about this particular TP-Link AP in this issue last year:

#https://github.com/socallinuxexpo/scale-network/issues/410#issuecomment-1197630016

nixinator, 2023-03-23:
if i had may way, i'd put a few of these in every room, especially the training rooms and actually workshop, rather than 'talks'..which seem to have low bandwidth requirements.  https://www.crowdsupply.com/traverse-technologies/ten64  , one can but dream.

davidelang, 2023-03-23:
talks have surprisingly high bandwidth requirements, even when the speaker 
doesn't tell everyone to download some image :-)

interesting box, but not a switch and not an AP (although you could add a wifi 
module, but then you would need external antennas, not a great thing)

and at $700+ each, there would be worries about them walking off that we don't 
have with equipment in the $100 range.

David Lang

On Wed, 22 Mar 2023, Lee Hughes wrote:

> if i had may way, i'd put a few of these in every room, especially the training rooms and actually workshop, rather than 'talks'..which seem to have low bandwidth requirements.  https://www.crowdsupply.com/traverse-technologies/ten64  , one can but dream.

nixinator, 2023-03-23:
It can functional as AP, and take two wifi cards , for twice the fun. It can functional as switch or router. However the price is crazy right now..

but it's the only truly open piece of network hardware i know about.

There is another company doing a totally open wifi AP , i'll see if can find them and get some samples.

davidelang, 2023-03-23:
each room switch has the following minimum
1. uplink
2. speaker
3. AV computer
4. AV mixer
5. AV camera
6+ links to access points

we like to provision multiple ports for each purpose, just in case something 
gets damaged.

I don't consider this a switch, it's just a linux system with multiple 
interfaces. I've been doing that for decades (up to 26 gig interfaces in a 
single machine). The linux kernel is going to be the limiting factor on the 
throughput. This isn't going to be a problem when you are doing firewalling, but 
when you just need switching, it's a problem (and it's why openwrt routers that 
have converted to DSA and do not use the switch as an 'accelerator' have such 
poor performance

David Lang

  On Thu, 23 Mar 2023, Lee Hughes wrote:

> It can functional as AP, and take two wifi cards , for twice the fun. It can functional as switch or router. However the price is crazy right now..
>
> but it's the only truly open piece of network [...]

nixinator, 2023-03-24:
Yeah, according to https://www.crowdsupply.com/traverse-technologies/ten64/updates/10g-options-and-performance

single flow is limited to 3G , which aint to shabby, seeing as most clients will only be 1G max anyway.

Unless your doing 10G to 10G transfers.

This little thing only use 15W :-).

But yeah, the FPGA module for this might be a step in the right direction.... 

interesting stuff.

davidelang, 2023-03-24:
On Thu, 23 Mar 2023, Lee Hughes wrote:

> Yeah, according to https://www.crowdsupply.com/traverse-technologies/ten64/updates/10g-options-and-performance
>
> single flow is limited to 3G , which aint to shabby, seeing as most clients will only be 1G max anyway.

with what packet size? (normally such tests are done with jumbo packets...)

> Unless your doing 10G to 10G transfers.
>
> This little thing only use 15W :-).
>
> But yeah, the FPGA module for this might be a step in the right direction....
>
> interesting stuff.

Interesting box if I was wanting to build a firewall, but where we need a switch 
and some APs, it's just not the right tool for the job.

David Lang

owendelong, 2024-01-28:
Is this candidate still under consideration?

sarcasticadmin, 2024-01-29:
I believe we can close this as we have already green light the belkins RT3200/Linksys E8450 #532

closed 2024-01-29 by sarcasticadmin