socallinuxexpo/scale-network #595: Writing to many MicroSD Cards

MrHamel, 2023-03-15:
I came across this https://store.balena.io/products/etcher-pro through a Twitter post https://twitter.com/geerlingguy/status/1636108199479726081, an appliance that can write images to 16 devices at once.

While we try to flash images ahead of time, emergencies can happen. It is expensive, but would give back an enormous amount of time if this product or something like it was around. When the RPi PXE system is working, it would eliminate any need for this, but just throwing it out there.

sarcasticadmin, 2023-03-24:
@MrHamel thanks for raising this up. Its absolutely something to be aware of!

>  When the RPi PXE system is working, it would eliminate any need for this, but just throwing it out there.

Im putting my bets on this for the time being :pie:

owendelong, 2023-03-24:
I plan to pick one up anyway.

owendelong, 2023-09-13:
I have one. I will try to remember to bring it to the next work party.

closed 2023-09-13 by owendelong