socallinuxexpo/scale-network #596: enable dhcpv6 server for kea on core machine

sarcasticadmin, 2023-03-23:
## Description

Relates to: https://github.com/socallinuxexpo/scale-network/pull/536

Kea config is currently only v4. Need to add our v6 version of the config.

## Acceptance Criteria
- Kea service running on core provides dhcpv6 server

nixinator, 2023-03-23:
Just our of curiosity, i thought V6 address were big enough so you don't need dhcp at all, and one less service is one less thing to go wrong?

davidelang, 2023-03-23:
we are passing some parameters via dhcp, so those are still needed.

David Lang

On Wed, 22 Mar 2023, Lee Hughes wrote:

> Just our of curiosity, i thought V6 address were big enough so you don't need dhcp at all, and one less service is one less thing to go wrong?

sarcasticadmin, 2023-03-23:
@nixinator Yes @davidelang is correct. Ultimately well we passing the v4 options via v6.

IIRC @owendelong had some other use cases for this as well

owendelong, 2023-03-23:
Yes... Sometimes you want a particular host to consistently get the same address. While EUI-64 sort of allows this, it's no longer default on the vast majority of systems. Using DHCP6 reservations provides a consistent way to manage these addresses.

PR #671 (2024-03-04): [READY] - kea dhcpv6 config and service
## Description of PR

Fixes: #596
Depends on: #670 (Will have to rebase after)

Kea DHCPv6 using all of the existing vlans from switch configurations

A lot of this was made very easy by previous work (thanks @kylerisse ): https://github.com/socallinuxexpo/scale-network/pull/252 ,inventory.py ipv6 support, and the retired isc-dhcpv6. This was all very useful as reference for this implementation.

## Previous Behavior
- kea didnt have dhcpv6
- lease time was set to 4000 sec for dhcpv4

## New Behavior
- kea dhcpv6 configuration ([dhcp6-server.conf.txt](ht [... cut]
closed 2024-03-04 by owendelong