socallinuxexpo/scale-network #597: PoE for sign Pis?

davidelang, 2023-03-23:
it was pointed out to me that if we PoE the sign Pis, then when we need to reboot them all, we should be able to power cycle them by toggling PoE on the switches.

owendelong, 2023-03-23:
Yes, I'm not at all opposed to this idea, so long as we implement it using Proper Pi PoE hats (that don't have the original issues).

nixinator, 2023-03-24:
PI are notoriously sensitive to dodgy power (low volts, amps, noise) etc etc.

I hoping that long POE runs don't cause havoc. Also what the power budget on the junipers...? can they actually run that many pi's over POE... do the Junipers need PSU upgrades to do this if they can?

davidelang, 2023-03-24:
These junipers have a rather large power budget (740w), their limit is that they 
only do af (18w per port)

Thinking through out layout, I would be surprised if any switch had more than a 
dozen PoE devices run off of it.

I tried to test long cable PoE on sunday, but the AP I plugged in wasn't getting 
the proper upstream connection, and I ran out of time to test.

I would welcome suggestions on how to test PoE capability over long cable runs, 
we'll do it at the next work party.

According to the Juniper stats, the APs were mostly drawing under 7w each.

The Pi4 4G devices we are using list a 5v 3A (15w) power supply, much of that is 
to support USB devices, and it looks like they need <10w from tests that I can 
find.

so 5v 2.4A devices like 
https://www.amazon.com/dp/B08J472HT2?psc=1&ref=ppx_yo2ov_dt_b_product_details 
should work (and I have used similar devices to power my raspberry pi 400 and 
orange pi 800 without any problems)

David Lang

On Thu, 23 Mar 2023, Lee Hughes wrote:

> PI are notoriously sensitive to dodgy power (low volts, amps, noise) etc etc.
>
> I hoping that long POE runs don't cause havoc. Also what the power budget on 
> the junipers...? can they actually run that many pi's over POE... do the 
> Junipers need PSU upgrades to do this if they can?

owendelong, 2023-03-24:
I realize the hát cost more, but they do have advantages. Where possible, I prefer we keep the PoE self-contained inside the unit being powered. With our current APs, that’s not possible, so dongles are the best remaining alternative. 

For much the same reason I didn’t want monitors with power bricks.

owendelong, 2023-11-09:
We can now get Juniper EX4200-48PX switches for about the same price as the EX4200-48P switches  we currently have.

They are nearly identical in every way, except that the PX models have a different color painted on the faceplate and will also support 802.3at (and a little more -- up to 30W per port). They have the same total 740W PoE power budget as the 48P, so they can only deliver a full 30W to about half of the ports, but we're nowhere near that on any of our switches.

Since they are nearly identical in price, my plan is to slowly (probably about 3 per year) phase these in as replacements for the 48P switches we currently have. (We seem to be losing about 1-2 switches per year currently to flash memory degradation, so I figure this will keep us ahead of that curve).

At some point during this process, the EX4300-48P (no PX in this line but P does 802.3AT (30W per port) and has a 950W total PoE budget). Potentially, if the price becomes similar, we might buy EX4300-48MP which gives two additional capabilities... First, all 48 ports support any of 100/1G/2.5G/5G/10GbaseT (10BaseT is unsupported, but I don't see that as an issue). Second, they support PoE++ up to 95 W per port.

Right now, the 4300-48P runs about $900@ and the 4300-48MP is running about $1200 (used), but they're also still being sold actively by Juniper. Eol was announced for 3/31/2024 on June 24, 2022. The replacement is the EX4100 Multigigabit (EX4100-48MP), but those remain very pricey ($6k+ used, $12K+ new).

sarcasticadmin, 2024-01-17:
As of the last work party: 1/13/24-1/14/24 we now have POE capabilities on all RPis

closed 2024-01-17 by sarcasticadmin