socallinuxexpo/scale-network #599: nix flake check / nix flake show error with "error: expected a derivation"

nixinator, 2023-03-24:
## Description
nix flake check
nix flake show

```
[nix-shell:~/src/scale-network]$ nix flake show
git+file:///home/l33/src/scale-network?ref=refs%2fheads%2fmaster&rev=280bc083828adec400c6f43f1493ee210acfbbf6
├───devShells
│   └───x86_64-linux
│       └───default: development environment 'nix-shell'
├───nixosConfigurations
│   ├───coreMaster: NixOS configuration
│   ├───coreSlave: NixOS configuration
│   ├───loghost: NixOS configuration
│   ├───massflash: NixOS configuration
│   └───signs: NixOS configuration
├───overlays
│   └───default: Nixpkgs overlay
└───packages
    └───x86_64-linux
        ├───scaleInventory: package 'scaleInventory'
error: expected a derivation

[nix-shell:~/src/scale-network]$ nix flake check
error: flake attribute 'packages.x86_64-linux.scaleTests' is not a derivation
(use '--show-trace' to show detailed location information)
```

Seems that something in the flake is not quite right, i think it maybe scaleTests.

## Acceptance Criteria
nix flake check
and
nix flake show

work.

sarcasticadmin, 2023-03-24:
you beat me to it! Closing #600 as a dup

sarcasticadmin, 2023-03-24:
This seems to be due to scaleTests being a new scope instead of an individual package: https://github.com/socallinuxexpo/scale-network/blob/280bc083828adec400c6f43f1493ee210acfbbf6/flake.nix Ideally it would be nice to not have to explicitly define each package in the scope.

Interesting discussions around doing this: https://discourse.nixos.org/t/namespacing-scoping-a-group-of-packages/13782/2

sarcasticadmin, 2023-11-08:
This has been fixed thanks to @Kranzes #625 

```
git+file:///home/rherna/workspace/scale-network?ref=refs%2fheads%2fmaster&rev=e857c20374465b3b1b810c70c8339a65286b4156
├───apps
│   ├───aarch64-linux
│   └───x86_64-linux
├───checks
│   ├───aarch64-linux
│   │   ├───openwrt-golden: derivation 'openwrt-golden'
│   │   ├───perl-switches: derivation 'perl-switches'
│   │   └───pytest-facts: derivation 'pytest-facts'
│   └───x86_64-linux
│       ├───core: derivation 'vm-test-run-core'
│       ├───loghost: derivation 'vm-test-run-loghost'
│       ├───openwrt-golden: derivation 'openwrt-golden'
│       ├───perl-switches: derivation 'perl-switches'
│       └───pytest-facts: derivation 'pytest-facts'
├───devShells
│   ├───aarch64-linux
│   │   └───default: development environment 'nix-shell'
│   └───x86_64-linux
│       └───default: development environment 'nix-shell'
├───formatter
├───legacyPackages
│   ├───aarch64-linux omitted (use '--legacy' to show)
│   └───x86_64-linux omitted (use '--legacy' to show)
├───nixosConfigurations
│   ├───coreMaster: NixOS configuration
│   ├───coreSlave: NixOS configuration
│   ├───loghost: NixOS configuration
│   ├───massflash: NixOS configuration
│   └───signs: NixOS configuration
├───nixosModules
│   └───bhyve-image: NixOS module
├───overlays
└───packages
    ├───aarch64-linux
    │   ├───massflash: package 'massflash'
    │   └───scaleInventory: package 'scaleInventory'
    └───x86_64-linux
        ├───massflash: package 'massflash'
        └───scaleInventory: package 'scaleInventory'
```

closed 2023-11-08 by sarcasticadmin

nixinator, 2023-11-19:
@Kranzes , thanks for that , you are da man!