socallinuxexpo/scale-network #601: Using Containers/Pods for Ease in Setup

MrHamel, 2023-03-25:
I am volunteering at non-profit this weekend that is hosting a LAN party, where they use special pods for the core + distro networking, and servers. It got me thinking, what if **everything** (switches + A/V + PDUs + WAPs, Ethernet + power cables, tape) was put into pods specifically made for each room like a ready to go kit that can be dropped off and picked up by anyone, setup + teardown would be even faster. It would eliminate the custom carts for ordinary flatbed carts.

Below are photos from the event, it is going on tonight and tomorrow, with teardown Monday. I can answer any questions about the equipment and get specifics about those containers.

https://drive.google.com/file/d/19iyZr5Z9jkshJ4wueLLoJaB2hAKmhdqN/view?usp=sharing
https://drive.google.com/file/d/1EBB60DI_oMYlp44rSUN9BzPgpHe4Geoy/view?usp=sharing
https://drive.google.com/file/d/1XxdCS7KY0mRL8lGd0eWTUnpIYleABFpE/view?usp=sharing
https://drive.google.com/file/d/1YaaGaql975aHy2b2z-U6ILKKf8ErapKc/view?usp=sharing
https://drive.google.com/file/d/1isDnW5Cwrosp31-9CuFdTm4eTVdPsnC6/view?usp=sharing
https://drive.google.com/file/d/1uSmoXyAB-bDnBvr8gWksUJVNA1_8yHEm/view?usp=sharing
https://drive.google.com/file/d/1xht7WzNjA01KkwyrUVmtZEeZfQqc57FX/view?usp=sharing
https://drive.google.com/file/d/1ycJxruZjD05bBLs6_x87idbPGdOKzslZ/view?usp=sharing

davidelang, 2023-03-25:
a few years ago we packed up with the APs in the room boxes, and when we had the 
work party, we had to go through every box to get the APs and their power 
supplies out so that we could flash them with updates.

That was enough extra work that this year we didn't put them in the room boxes.

There aren't enough rolls of tape to put them in every box, and when we are 
taping down, you want to have several rolls in use in the room at one time (and 
gaffer tape is not cheap, IIRC around $1k for the case which lasts ~2 years)

I really don't think that they store the cables in those nice portable racks, 
they probably have separate boxes for the cables.

boxes that are large enough and strong enough to hold the switches are going to 
be heavy, expenseive, and large. even stacked, will that many of them fit in 
storage? or would we need a larger storage unit? (remember, we never have more 
than one switch per room, so we would need 50+ of these containers, and some 
rooms have so many cables that they fill up the existing bin without the APs or 
switch)

with the current carts, moving things in and out of storage is pretty easy, roll 
the carts into the truck and strap them down. With flat carts and boxes like 
this, they would all have to be lifted onto the carts and off of the carts, and 
then strapped down. That requires a lot more time and muscle.

I can see how it would work well for other situations where storage is more 
flexible, there is paid (young) muscle to move things, and especially where 
there is a substantial amount of equipment to put in a given location, but I 
think it would not be a win for us.

David Lang

On Sat, 25 Mar 2023, Ryan Hamel wrote:

> I am volunteering at non-profit this weekend that is hosting a LAN party, where they use special pods for the core + distro networking, and servers. It got me thinking, what if **everything** [...]

nixinator, 2023-03-26:
Nice pictures.. That certainly looks like a lot of fun. Whats the hardware specs? Grab the configurations ;-).

It's much better to have things off the floor for noise, dust, heat etc etc etc. and it looks pro too. 

There are certainly good ideas there, I think the answer to our problems are IKEA, the swedish flat pack furniture manufacture taking the USA by storm at the moment and castor wheels.

https://wiki.eth0.nl/index.php/LackRack#:~:text=The%20LackRack%20is%20a%20stackable,local%20non%2Dcomputer%20hardware%20store.

Not only is the Lack Rack suitable for juniper hardware, it also doubles up a handy table, and avoids potential trip hazards. 

The addition of castor wheels to the legs means it's totally portable.

The tops of the of tables can have cool decals on them that say 'SCALE NOC' 'and a large colour picture of Robs face on them. 

As it's flat pack, it can be disassembled at the end and stored in a very compact manner. 

As I assembled the impossible to construct while boards, with their insane instructions *AND*  girlfriends in my life have put me me through the ikea 'assault' course, mainly on Sunday afternoons, assembling all manner of flat pack furniture....  I believe i have had the 'battle training' to achieve this task.

davidelang, 2023-03-26:
those tabletops are about twice as large as the juniper switches are, let alone 
the legs. so we would need over triple the space that we currently have the 
switches stored in (not to mention the extra setup/teardown time, and haivng to 
find space for the tables in every room)

As far as noise goes, solidly attaching the switches to something can make it 
into a sounding board, increasing the noise level. In many rooms we put the 
switches under the stage, so they are out of the way and the sound gets somewhat 
muffled by the skirt around the stage.

David Lang

On Sat, 25 Mar 2023, Lee Hughes wrote:

> Nice pictures.. That certainly looks like a lot of fun. Whats the hardware specs? Grab the configurations ;-).
>
> It's much better to have things off the floor for noise, dust, heat etc etc etc. and [...]

nixinator, 2023-03-26:
Well it seems my idea is somewhat Lack-ing, so we shall leave it at that.

owendelong, 2023-03-26:
> There aren't enough rolls of tape to put them in every box, and when we are 
> taping down, you want to have several rolls in use in the room at one time (and 
> gaffer tape is not cheap, IIRC around $1k for the case which lasts ~2 years)

$500/case, but by next year, that may well be $1k as it was about $280 last time
we purchased 3+ years ago.

Overall, I think that our hybrid solution where we have the cables and other things for each room pre-packaged in a room bin, where we can add the APs, AP PoE splitters, etc. is worth doing.

I don’t think having the switches in the room boxes is desirable.

It might be interesting to try an experiment where we unload all the carts into the upper floor of the conference center on Monday, then once APs are ready for distribution, the first step is to load all the accessories into the room boxes and then dispatch carts and teams to install rooms.

It would also be interesting to see if we could construct a fixture to more easily carry the IDF/MDF switches and routers, fiber, etc. for the initial IDF deployments.

davidelang, 2023-03-26:
On Sat, 25 Mar 2023, Owen DeLong wrote:

> It might be interesting to try an experiment where we unload all the carts 
> into the upper floor of the conference center on Monday, then once APs are 
> ready for distribution, the first step is to load all the accessories into the 
> room boxes and then dispatch carts and teams to install rooms.

would it really help to set the switches on top of the bins (they won't fit in 
the bins)?

we're already at the point of one cart per area (conf up, conf down, expo, 
ballroom), so (except for the APs) we can just wheel the carts around and drop 
off everything for each room at once

> It would also be interesting to see if we could construct a fixture to more easily carry the IDF/MDF switches and routers, fiber, etc. for the initial IDF deployments.

should not be a problem, how many switches do you want to move at once?

MDF needs to be carried down the steps no matter what, but there are 4x IDF + 
1x catwalk going to the expo building. Right now we have a bin for each with 
(most of) the cables used there this year.

the IDF switches are on the NOC cart (on the theory we may want them more easily 
for a work party or similar), I think the IDF bins are all on the expo cart .

where do each of the routers go?

I can see a short (4-5u) rack that is designed for a dolly to easily pick it up 
that would then allow you to stack bins on top of it securely (probably two wide 
to keep them from getting too tall). I could easily see this doing 4 bins and 4 
switches, possibly going to 5 switches and 6 bins (the stack would be a little 
under 5' tall and just under 3' wide, narrow enough to fit through a door, but 
barely)

we could build a custom flat cart that would hold this and disassemble to store 
in the large cart like the existing ones

David Lang

owendelong, 2023-03-26:
> It might be interesting to try an experiment where we unload all the carts 
> into the upper floor of the conference center on Monday, then once APs are 
> ready for distribution, the first step is to load all the accessories into the 
> room boxes and then dispatch carts and teams to install rooms.

would it really help to set the switches on top of the bins (they won't fit in 
the bins)?
That’s not what I’m advocating. Just placing APs, pi’s, and power supplies (splitters) in the bins. 
we're already at the point of one cart per area (conf up, conf down, expo, 
ballroom), so (except for the APs) we can just wheel the carts around and drop 
off everything for each room at onceYes. 

> It would also be interesting to see if we could construct a fixture to more easily carry the IDF/MDF switches and routers, fiber, etc. for the initial IDF deployments.

should not be a problem, how many switches do you want to move at once?There are a total of 4 IDF switches and 2 routers plus one server that get deployed throughout expo. That involves 5 fiber patches, 2-5 copper patch cables (depending on how PCC chooses to do certain handoffs), 7 IEC power cords, two router power supplies. 

MDF needs to be carried down the steps no matter what, but there are 4x IDF + 
1x catwalk going to the expo building. Right now we have a bin for each with 
(most of) the cables used there this year.Yes, there is no point in doing this for the CF IDF. This would just be for Expo. However, having separate bins for the IDFs doesn’t make much sense to me. They are the first things we deploy (ideally on Monday). The MDF has to be deployed first. The CF IDF is second priority, but generally due to the PCC logistics, it makes sense to deploy all of EX IDFs at the same time as MDF. Either in parallel with someone doing CF IDF or before. I believe currently, there aren’t bins specifically for the IDFs, but bins for the rooms containing them (A, F).

the IDF switches are on the NOC cart (on the theory we may want them more easily 
for a work party or similar), I think the IDF bins are all on the expo cart .It doesn’t really make sense given that we have one person (or at best one team) deploying the IDFs. Being able to quickly load those switches, routers, server, and all associated cables onto one of the small carts would much better facilitate the typical deployment. The fixture would make it easy to organize and easy to pull the switches out in whatever order they get deployed (vs having them stacked in a wagon). 

where do each of the routers go?2 in EX-MDF and one in CF-IDF. 

I can see a short (4-5u) rack that is designed for a dolly to easily pick it up 
that would then allow you to stack bins on top of it securely (probably two wide 
to keep them from getting too tall). I could easily see this doing 4 bins and 4 
switches, possibly going to 5 switches and 6 bins (the stack would be a little 
under 5' tall and just under 3' wide, narrow enough to fit through a door, but 
barely)I was more thinking a small plywood frame which switches could slide into vertically face up with side pegs to hang cooled fiber and patch cords. 

we could build a custom flat cart that would hold this and disassemble to store 
in the large cart like the existing onesI don’t think that’s needed. I think the existing plastic carts do the job. Owen

David Lang

davidelang, 2023-03-26:
On Sun, 26 Mar 2023, Owen DeLong wrote:

> On Sat, 25 Mar 2023, Owen DeLong wrote:

> That’s not what I’m advocating. Just placing APs, pi’s, and power supplies (splitters) in the bins. 

Ok, when you said all, I thought you were including switches as part of 'all' 
:-)

right now we use a mix if PoE and wall warts to power these, if we go to all PoE 
(and figure out why we seemed to have some problems with PoE in some places) we 
could do this.

In the past it has seemed to make sense to have one person going around placing 
the APs and dealing with their power, but that could just be mixing cause and 
effect.

>> It would also be interesting to see if we could construct a fixture to more easily carry the IDF/MDF switches and routers, fiber, etc. for the initial IDF deployments.
>
>> should not be a problem, how many switches do you want to move at once?

> There are a total of 4 IDF switches and 2 routers plus one server that get 
> deployed throughout expo. That involves 5 fiber patches, 2-5 copper patch 
> cables (depending on how PCC chooses to do certain handoffs), 7 IEC power 
> cords, two router power supplies. 

>> MDF needs to be carried down the steps no matter what, but there are 4x IDF + 
>> 1x catwalk going to the expo building. Right now we have a bin for each with 
>> (most of) the cables used there this year.

> Yes, there is no point in doing this for the CF IDF. This would just be for
> Expo. However, having separate bins for the IDFs doesn’t make much sense to
> me. They are the first things we deploy [...]

we do have separate boxes for each of the 4 IDFs with all the small cables 
needed for that IDF. we can change this at the work party as we think 
appropriate (I was planning to put a few extra cables in each bin in any case, 
with the idea that the bin stays in the IDF for the show)

>> the IDF switches are on the NOC cart (on the theory we may want them more 
>> easily for a work party or similar), I think the IDF bins are all on the expo 
>> cart.

> It doesn’t really make sense given that we have one person (or at best one
> team) deploying the IDFs. Being able to quickly load those switches, routers,
> server, and all associated cables onto one [...]

no disagreement, this was just saying where they are currently.

>> where do each of the routers go?

> 2 in EX-MDF and one in CF-IDF. 

is EX-MDF the same as what I'm thinking of as SW-IDF? (the room by the loading 
dock)

> I can see a short (4-5u) rack that is designed for a dolly to easily pick it
> up that would then allow you to stack bins on top of it securely (probably
> two wide to keep them from getting too [...]

> I was more thinking a small plywood frame which switches could slide into 
> vertically face up with side pegs to hang cooled fiber and patch cords. 

I think there are a LOT more cables than would work reasonably for this. If we 
were just talking about the cables needed to connect the switches to the MDF I 
think your idea would be reasonable, but if we add all the cables for the 
patches from the IDF switches elsewhere, I think there are a lot more of them (I 
think about a dozen for the NW IDF). We can check how many are in each bin at 
the work party. I agree that the standard size bins are overkill, I just don't 
know if it's worth having a different size bin for this purpose (and as you say, 
it's a good question if they should be a separate bin per closet or not)

I would also question if the cables would stay on the pegs with the trip over 
the bricks or if they would end up bouncing off (in part, depends on the peg 
angle

whhen I say 'short rack', I'm thinking plywood box, but I was thinking of the 
switches being horizontal in it rather than vertical, so that the bins could be 
stacked on top and still have access to the switches. Close to the same thing, 
just used in a different way.

This sort of box could be used to hold the servers in place in the cart a bit 
more solidly as well, so I think it's a good idea no matter what. I'll build at 
least the start of one.

David Lang

owendelong, 2023-09-13:
> > There are certainly good ideas there, I think the answer to our problems are IKEA, the swedish flat pack furniture manufacture taking the USA by storm at the moment and castor wheels.

I prefer better weather. Let's avoid the storms from Sweden.

> https://wiki.eth0.nl/index.php/LackRack#:~:text=The%20LackRack%20is%20a%20stackable,local%20non%2Dcomputer%20hardware%20store.

While I appreciate the ingenuity, no... Just no. This really doesn't fit our needs. Actually on the floor and under the stage is virtually ideal in terms of noise reduction in our environment.

We like things to last and most IKEA furniture is made of MDF and/or Particle board which means you only get to assemble/dissassemble it once or twice before the mating points start disintegrating, the screws strip out, and the whole thing fails catastrophically.

Hard pass on the Swedish crap factory.

owendelong, 2024-01-28:
I think this thread has run its course and other than the IDF cart/container (which is captured in another thread) there's no there there. As such, I'm closing this issue as WONTFIX.

closed 2024-01-28 by owendelong