socallinuxexpo/scale-network #602: additional purchases for 2024

davidelang, 2023-03-30:
things we should purchase for 2024

Scott W has a list of items to purchase

In addition

I would like to buy 1 more monitor, we deployed 19 of the 20 this year and would have deployed another one or more if we had the additional Pis

we should purchase a few more Pis when they become more available later this year.

nixinator, 2023-03-30:
A remote control teleprocessing robot would be good. So i can inspect patch panels remotely.

davidelang, 2023-03-30:
longer ac power cables (odd color for identifcation)

more storage bins

MrHamel, 2023-03-30:
https://www.gaffgun.com/ - While not necessary, this can be useful in speeding up setup.

davidelang, 2023-03-30:
we've talked about getting some gaffguns in the past, what do they cost?

MrHamel, 2023-03-30:
$250/pop - Varies based on cable guide size, catering to the number of cables being taped down.

https://www.gaffgun.com/products/gaffgun?variant=1261982524

davidelang, 2023-03-30:
consider PoE for sign Pis so they can be restarted remotely.
external splitters are ~$9 each
internal boards are ~$30 and may need new cases. We would also need to add this to all Pis rather than just getting enough external splitters for the sign Pis

davidelang, 2023-03-30:
@MrHamel how many would we need (since we split up and cable multiple rooms at a time)

MrHamel, 2023-03-30:
Depends on the average number of people participating in installs, a dozen would get expensive so it depends on physical health being a concern for volunteers. The young and more fit volunteers can get away with not using them, but for my knees and ankles, it would help me in participating with setup next year.

davidelang, 2023-03-30:
another toner kit

sbibayoff, 2023-03-30:
There is really only 3 areas where we have long runs of cable that need to
be taped down (not along a wall), lower conf, ball d-e, and reg. And
considering those are set up at different times, 2 or 3 would be more then
enough.

davidelang, 2023-03-30:
a couple rangefinders

davidelang, 2023-03-30:
more ram for the server

davidelang, 2023-03-30:
replacement switch for at least the one that has lost it's flash (how can we test this at the work party and find if we need to replace them)

davidelang, 2023-03-30:
see #108

nixinator, 2023-09-10:
Can we rent a few electric scooters, because sometimes you just need something in building A / B quickly, either equipment or inspection.

Zooming (well carefully riding) might save a bit of shoe leather.

davidelang, 2023-09-10:
On Sat, 9 Sep 2023, Lee Hughes wrote:

> Can we rent a few electric scooters, because sometimes you just need something in building A / B quickly, either equipment or inspection.
>
> Zooming (well carefully riding) might save a bit of shoe leather.

that's why I bring my bike :-) (which is available for anyone else to use who 
needs to zip over somewhere)

sit down electric scooters are rather pricy to rent (we have done so a couple 
times in the past when we had people who needed them) I don't remember the exact 
cost, but I think it was several hundred if not over $1k

David Lang

owendelong, 2023-09-13:
> longer ac power cables (odd color for identifcation)
> 
Agreed.

> more storage bins

Let's inventory what we have... Last I looked, I thought we had a lot of empty storage bins and not a lot of empty shelf space.

owendelong, 2023-09-13:
> consider PoE for sign Pis so they can be restarted remotely.
> external splitters are ~$9 each
> internal boards are ~$30 and may need new cases. We would also need to add this to all Pis rather than just getting enough external splitters for the sign Pis

I do like the tamper resistance of the internal boards. Personally, I'd like to go to solid aluminum cases for Pis going forward anyway.

owendelong, 2023-09-13:
> $250/pop - Varies based on cable guide size, catering to the number of cables being taped down.
> 
> https://www.gaffgun.com/products/gaffgun?variant=1261982524

I will push to purchase two of these for this year...I think that will be sufficient.

owendelong, 2023-09-13:
> a couple rangefinders

Can you provide a link to what you mean here?

owendelong, 2023-09-13:
> more ram for the server

We may simply replace the server that has the RAM issue.

owendelong, 2023-09-13:
> replacement switch for at least the one that has lost it's flash (how can we test this at the work party and find if we need to replace them)

I intend to order an EX4200-48PX ($170) from Hula to replace that one, but I figured I'd see how many we need after cycling all of them at the work party. Currently EX4300s are still about $1,000, but that's down from $2,500 a year ago, so if they are under $500 next year, I'll be pushing to consider using them to replace failed units starting at that point. The 4300 offers significant advantages over the 4200, including up to 4x10G uplinks (where we need fiber connections). Further the 4300 is not yet EOL and current JunOS is still issued for it, though with the 4400s out, the writing is on the wall.

owendelong, 2023-09-13:
> Can we rent a few electric scooters, because sometimes you just need something in building A / B quickly, either equipment or inspection.
> 
> Zooming (well carefully riding) might save a bit of shoe leather.

A couple of Razor style scooters might make sense. I'm not sure electric is necessary (or even all that useful).
Riding scooters are far to expensive to deploy absent medical need.

owendelong, 2023-09-13:
> that's why I bring my bike :-) (which is available for anyone else to use who needs to zip over somewhere)

Your bike is not usable by 99% of the NOC staff due to the height discrepancy. Even with the seat adjusted to its lowest possible setting, your frame size is too much for most of us.

owendelong, 2023-09-13:
I think that using an issue to track this is horribly ill-suited for the task. I will set up a Google sheet and share the link here.

MrHamel, 2023-09-13:
> Caution: This is an external email and may be malicious. Please take care when clicking links or opening attachments.

Owen,

EX4300-48P's are below $500 on eBay, although that is without the EX-UM-4X4SFP module. As those modules are stupidly expensive when they do show up on eBay, it would make more sense to get the appropriate 40G QSFP's from FS, or another vendor willing to donate some parts -- with NANOG coming up next month that is certainly a possibility. Yes, the fiber capacity is definitely overkill, but I'm just going by price, currently readily available parts, and it definitely future proofs some infrastructure.

Let me know your thoughts.

owendelong, 2023-09-13:
I’d want to buy from a reputable used hardware vendor rather than eBay. I’ve had very mixed results with devices purchased via eBay.

But that’s a good sign that the reputable used market will likely get there sooner than later.

Owen

owendelong, 2023-11-09:
> consider PoE for sign Pis so they can be restarted remotely. external splitters are ~$9 each internal boards are ~$30 and may need new cases. We would also need to add this to all Pis rather than just getting enough external splitters for the sign Pis

Thinking about this more, I come back to favoring the internal PoE hats for all PIs. Eliminating external power bricks has strong appeal to me and there's no reason we couldn't power all Pis with PoE. Since we seem to have decided against considering NanoPis (which don't support an internal hat), Further, they're more like $20 in the open market instead of $30.

davidelang, 2023-11-09:
I believe that you were also considering new cases for the Pis (since these have 
ended up being a bit fragile)

Can we get them in in the next week for this work party? or is this something 
for the next work party?

David Lang

On Wed, 8 Nov 2023, Owen DeLong wrote:

> Thinking about this more, I come back to favoring the internal PoE hats for all PIs. Eliminating external power bricks has strong appeal to me and there's no reason we couldn't power all Pis with [...]

owendelong, 2023-11-10:
I’ll see what I can do. How many cases do we need? I might resort to laser cutting acrylic cases stacked together with machine screws.

Not the easiest things in the world to assemble, nor the prettiest option, but highly functional and very durable.

Owen

sarcasticadmin, 2023-11-18:
Workparty Conclusion:

### To Order
- 2 AOC Monitors
- 3 pi 5s - 8GB Preferred with Fan
- Acrylic Cases for Pis - @owendelong to build these out
- 27 POE Hats for Pi 4
- 1 Toner Kit - Currently only have 1 toner in the inventory
- Laser Ruler - 4x, 100M
- Belkin RT3200/Linksys E8450 - Order as many new APs as budget allows (Need ~120 ulimately)

### Got
- Longer AC  Cables (odd colors for identifications) - @davidelang 
- Replacement 4200 (PX) - 2x

## Not Needed
- More bins for storage:
  - Already have extra empty bins
  - Open slots remain for new bins open
- RAM - Going to get newer server hardware instead

### Next Year (https://github.com/socallinuxexpo/scale-network/issues/634)
- Gaffgun - 2x with simplest  package for 2inch gaffer tape (~$250 per)
- Liq 100 Fluke - Link Runner

### Outstanding Questions
- Whats Scotts List for Cabling?

closed 2023-11-18 by sarcasticadmin