socallinuxexpo/scale-network #617: Nix openwrt builder (uses upstream buildimage from openwrt)

nixinator, 2023-09-09:
## Description
relates to #611

Nix seems to have image builder for openwrt (it uses just uses nix to  hotwire the image builder) .

https://github.com/astro/nix-openwrt-imagebuilder

I got it to build an image , however i can't test it.

@sarcasticadmin , it may have value, it may not.... wadda ya think?

sarcasticadmin, 2023-09-10:
@nixinator definitely would be interested in looking into this.

Share with me the image you built and I can run it on our hardware fairly easily. `magic-wormhole` whenever youre ready :slightly_smiling_face:

sarcasticadmin, 2024-01-25:
This is going to require a lot more work since it looks like the existing project cant build from source: https://github.com/astro/nix-openwrt-imagebuilder/issues/24#issuecomment-1484081593

sarcasticadmin, 2024-02-10:
Another potent alternative that @MatthewCroughan found: https://www.liminix.org

They also have some recent updates regarding the Belkin RT3200: https://www.liminix.org/2023/11/23/arms-length.html

MrHamel, 2026-02-12:
Closing this in favor of the mixos implementation happening on the OpenWrt One's.

sarcasticadmin, 2026-02-13:
mixos is not a foregone conclusion. It looks promising and 23x is going to be a proving ground. Well keep this open for now