socallinuxexpo/scale-network #618: Aruba AP25 Access Points as candidate replacement

owendelong, 2023-09-13:
## Description
Aruba AP25s are not super cheap ($300@), but up to 25 they will elect a controller amongst themselves. In our environment, I believe the best deployment would be two controllers (one in each building).

I'm using these now both at home and in my office and I'm pretty impressed with:

1. Ease of administration
2. Efficiency and Effective operation
3. Performance (802.11ax clients see upwards of 600Mbps down and as much as 800Mbps up real world over AT&T Symmetrical Gig fiber) These numbers are close to what I get from a laptop plugged directly into the AT&T router on a wired interface.

Downsides:

1. Very much not open source
2. Controllers aren't cheap and neither are the APs
3. Fairly limited visibility/diagnostics due to super simplified UI
4. Can be configured to operate without an untagged VLAN (though this must be done prior to adoption into the site)
5. PoE Native

## Acceptance Criteria
None specified as yet. Just tossing this up as a candidate for discussion and consideration. I, myself, am of mixed emotions about this in our environment. If we do decide that the convenience and capabilities are worthy of consideration, I'd be willing to try and approach Vendors and/or Aruba/HP about trying to get equipment donated or provided at a reduced cost.

nixinator, 2023-09-24:
they look cool.

https://www.arubainstanton.com/products/access-points/access-point-25/

100 max clients.

shame the don't support the new 6ghz and 7ghz wifi 6 bands, but i doubt many client devices do at the moment.

i'd be interested to see what there channel planning and floor mapping software is like, and if they can hand off distant clients onto better AP's.

Roaming is always hit and miss with AP's, and with wifi clients... but maybe they have magic in them.

davidelang, 2023-09-24:
100 max clients isn't good.

we have many rooms with more people that would push that limit with just one 
device per person (although if it's 100 on 2.4GHz and 100 on 5GHz that may 
handle it)

Also not thrilled with moving away from opensource on the APs

David Lang

On Sun, 24 Sep 2023, Lee Hughes wrote:

> they look cool.
>
> https://www.arubainstanton.com/products/access-points/access-point-25/
>
> 100 max clients.
>
> shame the don't support the new 6ghz and 7ghz wifi 6 bands, but i doubt many client devices [...]

owendelong, 2023-11-09:
Going to go ahead and close this.

closed 2023-11-09 by owendelong

ptrlow1, 2023-11-09:
Agree with David. Not being open source rules it out.
Also found out that the AP25 uses a broadcom soc, which don't work with OpenWrt

The OpenWrt wiki has this statement:
> Devices with Broadcom WiFi chipsets have limited OpenWrt supportability (due to limited FLOSS driver availability for Broadcom chips). Consider this when choosing a device to buy, or when deciding to flash OpenWrt on your device because it is listed as supported. See [Broadcom WiFi](https://openwrt.org/meta/infobox/broadcom_wifi) for details.

nixinator, 2023-11-16:
What a disappointing effort from broadcom.