socallinuxexpo/scale-network #627: Remove dependencies on test and build containers

sarcasticadmin, 2023-10-07:
## Description

Spun out from: https://github.com/socallinuxexpo/scale-network/issues/418#issuecomment-1523635591
 
> However, to echo @kylerisse other comments earlier in the thread, I really dont think we need the containers we have.
> 
>     ansible-tester - Most of this can be done via nix. We were largely using the container for consistency for the packages.
>     scale-signs - Again were largely using the container for consistency for the packages, even if we have to use it "as-is" we can get the correct packages on the vm directly via nix now.
>     scale-openwrt - Ditto for this as well
> 

## Acceptance Criteria
- Remove the dependencies on the build and tests containers in this repo

owendelong, 2024-01-28:
What else is needed to finish this?

sarcasticadmin, 2024-01-29:
We can close this given that the majority of the work has been done and stable. The latest remaining container in the build process will be dealt with as part of #617

closed 2024-01-29 by sarcasticadmin