socallinuxexpo/scale-network #632: Work Party Nov. 18-19 2023 - Checklist

sarcasticadmin, 2023-11-08:
## Description

```
Null Space
2522 N Ontario St, Burbank, CA 91504

Nov. 18-19, 2023 9AM-Dark Pacific Time
```

Items:
- [ ] Test New Trailer for moving our gear
- [ ] Review 21x Supply Requests: https://github.com/socallinuxexpo/scale-network/issues/602
- [ ] Check the Gaffer Tape Inventory
- [ ] Finalize decision around purchasing new hardware (belkin 3200) #532 
- [ ] Repack all cables (ideally in sets by room, but by length if necessary)
> Feel free to update with anything Ive missed

## Acceptance Criteria
- Have a productive Work Party

sarcasticadmin, 2023-11-18:
Gaffer tape -  2 year supply
Repack all cables (ideally in sets by room, but by length if necessary) - Already done from 21x cleanup

sarcasticadmin, 2023-11-28:
@sbibayoff @owendelong @davidelang were did we land with the  `Proper labels on boxes via labeller`? I feel like that was pretty much wrapped but by sunday

Overall I think we had a very productive work party! Thank you to all who were able to participate!

owendelong, 2023-11-28:
I think there is still work to do, but much progress was made.

Owen

closed 2024-01-17 by sarcasticadmin