socallinuxexpo/scale-network #633: How to indicate latest flash on Belkin RT3200

sarcasticadmin, 2023-11-08:
## Description

From the discussion https://github.com/socallinuxexpo/scale-network/issues/532#issuecomment-1684553673

```
True, but they are both triple LEDs capable of rendering multiple colors, therefore we could use different colors of internet LED (or even power LED) potentially for conference version signaling.

I’ll note that on the RT3200, the factory firmware uses the power LED to indicate WPS pairing enabled.

Owen
```

## Acceptance Criteria
- Identify the best way to visually indicate the current build
- Implement the change via the LED mechanisms in Openwrt

owendelong, 2023-11-08:
Unless anyone has better ideas, I propose the following color code for the "Internet LED":

Year Ending:                Color:
0                                     Blue (0x0000ff)
1                                     White (0xffffff)
2                                    Amber (0xffc000)
3                                    Green (0x00ff00)
4                                    Red (0xff0000)
5                                    Cyan (0x00ffff)
6                                    Magenta (0xff00ff)
7                                     Yellow (0xffff00) -- close to Amber, but if 5 year old firmware, obvious in other ways.
8                                     Pink (0xff8080) -- we should make sure this is easily distinguishable from red and magenta
9                                     Bright Green (0x80ff80) -- We should make sure this is easily distinguishable from green

Further, I propose the following for the power LED:
Power not good: Black (0x000000) -- Likely we can't change this if the power isn't on. ;-)
Power good, internet unreachable -- Red (0xff0000)
Power good, internet available, 2.4g OK, 5g down -- Yellow (0xffff00)
Power good, internet available, 2.4g down, 5g OK -- Magenta (0xff00ff)
Power good, internet available, 2.4g OK, 5g OK -- White (0xffffff)

Feedback welcomed and encouraged.

sarcasticadmin, 2023-11-08:
Thanks @owendelong Im good with this proposal.

The only thing I would want to confirm is that it doesnt conflict with the existing LED states of the AP. Things like `ready to flash` or `flash failed` are the ones that come to mind. Typically in these states the LED is blink so maybe its not an issue though

owendelong, 2023-11-08:
Wouldn't context be sufficient for this as well? We know when we're flashing or trying to flash something, so we wouldn't expect to see those blinking states in the wild and they'd be pretty distinctive when we are flashing, I would think.

Since we wouldn't take control of the LEDs to represent our intended state until our flashed-on firmware is loaded and running and networked, I don't think there's a conflict. Let me know if you think I'm missing something important.

It does occur to me that I didn't have a state to represent Power good, internet available, all radios down. Perhaps we should make that Yellow and make Power + 2.4G OK, 5G down Green (0x00ff00) instead.

davidelang, 2023-11-08:
I don't think there will be many cases where we have one or the other band down, 
so I think all radios up or all radios down is sufficient.

that frees up a slot for other purposes

David Lang

On Wed, 8 Nov 2023, Owen DeLong wrote:

> Wouldn't context be sufficient for this as well? We know when we're flashing or trying to flash something, so we wouldn't expect to see those blinking states in the wild and they'd be pretty [...]

owendelong, 2023-11-08:
I don't think there's a "free slot" issue here. We have at least 10 easily distinguishable colors (likely considerably more) on each of 2 LEDs... If we're really worried, that gives us at least 100 different meanings we could convey, but I really don't want to make this that complex.

However, if we want to indicate other things beyond what I've proposed for power, I don't think we need to give up the individual radio indications to do so.

sarcasticadmin, 2025-02-25:
Closing since the Belkin 3200 mt-7622 has been superseded by the Openwrt One mt798x

closed 2025-02-25 by sarcasticadmin