socallinuxexpo/scale-network #634: Future Purchase Items for SCaLE 22x

sarcasticadmin, 2023-11-18:
## Description

Items that weve pushed from SCaLE  21x to SCaLE 22x:

- Gaffgun - 2x with simplest  package for 2inch gaffer tape (~$250 per) https://www.brontapes.com/GaffGun
- Liq 100 Fluke - Link Runner

## Acceptance Criteria
- Ordered items

sarcasticadmin, 2024-10-10:
Looks like `MC-1500-595-WT-BK` was superseded see the page: https://www.bradyid.com/labels/aggressive-adhesive/bmp51-bmp53-indoor-outdoor-vinyl-labels-cps-brus-11881?part-number=mc-1500-595-wt-bk Its been replaced with the `M5C-1500-595-WT-BK`: https://www.bradyid.com/labels/permanent-adhesive/all-weather-permanent-adhesive-vinyl-label-tape-ribbon-for-m4-m5-printers-cps-4647839?part-number=m5c-1500-595-wt-bk

owendelong, 2025-02-02:
We're getting close to the point where I don't trust orders to get done in time for product to arrive before show. Have orders been approved/placed? How are we on Gaffer Tape? It might be time to order more of that IIRC.

davidelang, 2025-02-02:
we recevived all the cables, I purchased the two power strips, we have more than 
half a box of gaffer tape.

David Lang

On Sat, 1 Feb 2025, Owen DeLong wrote:

> Date: Sat, 01 Feb 2025 21:14:43 -0800
> From: Owen DeLong ***@***.***>
> Reply-To: socallinuxexpo/scale-network
>     ***@***.***>
> To: socallinuxexpo/scale-network ***@***.***>
> Cc: Subscribed ***@***.***>
> Subject: Re: [socallinuxexpo/scale-network] Future Purchase Items for SCaLE
>     22x (Issue #634)
> 
> We're getting close to the point where I don't trust orders to get done in time for product to arrive before show. Have orders been approved/placed? How are we on Gaffer Tape? It might be time to order more of that IIRC.
>
>
>

owendelong, 2025-02-22:
Where do we stand on the label tapes and are we finally actually getting gaff guns or kicking that down the road again?
What about the link runner?

sarcasticadmin, 2025-02-23:
The following Items from the list have been purchased:

- All Weather Permanent Adhesive Vinyl Labels for M5 Printers - 1.5", Black on White 	$77.99 	2 	$155.98
Catalog #: M5C-1500-595-WT-BK

- Self-Laminating Aggressive Adhesive Vinyl Wrap Around Labels with Ribbon for M4 M5 Printers - 2.5 " 	$58.99 	1 	$58.99
Catalog #: M4-51-417

- ScotchBlue Painter's Tape, 0.94" Width x 60 Yards, 8 Rolls, Blue, Multi-Surface, Indoor/Outdoor, 45% Renewable Resources, No Residue, DIY, Glass, Wood, 0.94 inches, Made in USA or Imported

> Where do we stand on the label tapes and are we finally actually getting gaff guns or kicking that down the road again?
> What about the link runner?

Pushing to 2026

owendelong, 2026-01-15:
Can this be closed now that 22x is distant in the rear view?