socallinuxexpo/scale-network #635: New server hardware

sarcasticadmin, 2023-11-18:
## Description

Our existing server hardware is as follows:

- Motherboard: X8DTH-6F
- CPU: ~Xeon 2011 2x
- RAM: 48G DDR
- Harddrives: 2TB SAS 7K 4x

## Acceptance Criteria
- Getting 2 servers that are ~2018 CPUs

nixinator, 2023-11-19:
as were doing quite a bit more nix, something with quite a few cores could be useful and a bit of memory..., if we need to use it a remote nix builder.

I wonder if @bcantrill can bring one of his swish new https://oxide.computer along for a stress test.

in the meantime ebay has lots of servers, unless you US folks know of some other way to get reasonbly priced server hardware?

sarcasticadmin, 2023-11-19:
We have a couple of avenues to explore for getting hardware. Anything we get will certainly have more horsepower, cores and ram that what we have at the moment.

owendelong, 2023-11-22:
Worst case, I have a pair of towers that are underutilized at home.

The specs may not exactly match, but they’ll be reasonably close:

CPU: model name	: AMD Ryzen 5 3600X 6-Core Processor
Disk: 2xNVME Samsung SSD 970 EVO Plus 1TB
	2xSpinning Hitachi HDS72202 2TB
RAM: 64GB (4x16GB DDR4-2133 Corsair CMW64GX4M2E3200C16)

If we don’t find an alternative, I can try and work out something with Ilan. I purchased these machines out of pocket primarily to do folding at home to help COVID research, with an intent to eventually repurpose them to lab equipment to learn and experiment with things like Hypervisors, Kubernetes, etc.).

I’ll end up buying new equipment to backfill eventually, so this isn’t ideal, but it is doable if we get desperate.

nixinator, 2023-11-22:
If we got good strong machines, we can nix them and configure them as remote builders, if we need to some patching and rebuilding.

Might be fun.

owendelong, 2023-11-22:
We really need 3 (possibly 4) servers…

We need two to run on-site at the show. These don’t have to be particularly powerful, but they do need enough RAM and CPU capability to run a hypervisor and VMs to provide basic services (KEA4, KEA6, RadvD (though with KEA6, it would simplify the Radv requirements to the point we could do those on the SRXs and the Expo switches), DNS, HTTP (signs, reg), and probably some other miscellaneous services I’m forgetting). It would be nice to be able to add (one of) NAT64/CLAT/464XLAT and the ancilliary DNS64 to support it. Ideally, we’d also want to add some form of data collection, alerting, and visualization (aka monitoring) to them.

IMHO, 6 cores @3Ghz+, 32GB RAM, and 2+TB disk is more than adequate for these. Ideally, each of the above capabilities would be in some form of self-contained portable appliance structure (container, VM, etc. set up to be deployed and redeployed from development environment to production with minimal effort and delay).

We’d also like to have a machine or two that can be used for development of the above production environments year round. My plan is to host that machine at Hurricane Electric in my cabinet. We might want more processor horse power on that machine to facilitate faster builds, especially if we are cross-compiling. We might also want a native ARM compile server (which can be done fairly easily).

It might be desirable to have a Dev and a Test copy of the above machine, but I’d settle for one to be used for Dev+Test for now.

Eventually, for the dev environment, it would also be nice to have a pair of R.Pi (4 or 5) powered over PoE and net-boot with an IPkvms attached to them, so that we could do remote testing of Sign and Reg processes.

The PoE would allow us to power cycle the Pi remotely and the IPkvms would allow us to remotely use the Pi for testing. I’d want a reliable immutable Linux on an SD card so that the Pi would boot from that if the netbook failed, to give us the ability to better diagnose things.

I know there’s a lot of enthusiasm around NIX here and that’s great, but I think it’s important to maintain perspective on what the goals are here and recognize that nix is just a tool to help us achieve those goals, not a goal in and of itself.

Owen

sarcasticadmin, 2023-11-22:
all sounds great, lets get our basis covered for the conference and then look beyond that.

sarcasticadmin, 2024-01-17:
We had a generous donation and now have the following hardware for utilization as the hypervisors:

- 1x - Lenovo ThinkSystem SR650 Type: 7X06
- 2x - IBM System x3650 M5 Type: 8871AC1

closed 2024-01-17 by sarcasticadmin