socallinuxexpo/scale-network #636: openwrt make build-img fails when ran without templates

sarcasticadmin, 2023-11-18:
## Description

For the openwrt build: `make build-img` fails without including `templates`  due to `files` missing in the root of the build dir for openwrt. Should be an easy fix but will confirm after working on the ubi image for the belkin 3200/e8450

https://github.com/socallinuxexpo/scale-network/blob/1e7d542d182b9a14cbcfd620d87afdd667167190/openwrt/Makefile#L95-L96

## Acceptance Criteria
- `make build-img` works without having the call `make templates` before hand

nixinator, 2023-11-23:
this might be an opportunity to move this builder over to the nix one , However, the nix one needs more testing.... maybe we can test it out and see if it can build images.

owendelong, 2024-01-28:
@sarcasticadmin Are we ready to spin up VMs on the Dev Server in my cabinet? This might be a good thing to spin up as a test VM for exactly that purpose.