socallinuxexpo/scale-network #650: investigate rperf to replace iperf3

tweak42, 2024-01-14:
## Description
iperf does not allow multiple concurrent client connections for load stress testing
rperf does support this feature
https://github.com/opensource-3d-p/rperf

## Acceptance Criteria
rperf working with multiple clients simultaneously
added to openwrt boot image

nixinator, 2024-02-01:
Hello @tweak42 ,

https://openwrt.org/packages/start?q=rperf&do=search

So it's going to be have to built and compiled into the image.

However, it's says it supports 'linux', but doesn't mention anything about arm based router boards.

Looks like an interesting iperf3 replacement, but it hasn't been touched for 2 years, which either means it's perfect or unmaintained.

sarcasticadmin, 2026-02-13:
Still interested in investigate this