socallinuxexpo/scale-network #652: BSS Coloring for Wi-Fi 6

MrHamel, 2024-01-14:
## Description
Add BSS Coloring ID based on the last octet of the management IP address (2-63)

## Acceptance Criteria
What is the expected outcome that resolves this issue?

BSS Coloring is added to the script (https://github.com/socallinuxexpo/scale-network/blob/master/openwrt/files/etc/udhcpc.user)

sarcasticadmin, 2024-01-14:
Example `/tmp/udhcpc.params`:

[udhcpc.params.txt](https://github.com/socallinuxexpo/scale-network/files/13932850/udhcpc.params.txt)

sarcasticadmin, 2024-01-14:
~# cat /etc/board.json | jq -r .
{
  "model": {
    "id": "linksys,e8450-ubi",
    "name": "Linksys E8450 (UBI)"
  },
  "led": {
    "wan": {
      "name": "WAN",
      "sysfs": "inet:blue",
      "type": "netdev",
      "device": "wan",
      "mode": "link tx rx"
    }
  },
  "network": {
    "lan": {
      "ports": [
        "lan1",
        "lan2",
        "lan3",
        "lan4"
      ],
      "protocol": "static"
    },
    "wan": {
      "device": "wan",
      "protocol": "dhcp"
    }
  },
  "wlan": {
    "wl0": {
      "path": "platform/18000000.wmac"
    },
    "wl1": {
      "path": "1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0"
    },
    "wl2": {
      "path": "1a145000.pcie/pci0001:00/0001:00:01.0/0001:01:00.0"
    }
  }
}
```

sarcasticadmin, 2024-02-04:
Additional info was recently added to the openwrt wiki regarding BSS coloring: https://openwrt.org/toh/linksys/e8450#beam_forming_and_enable_basic_service_set_bss_coloring

davidelang, 2024-02-05:
did you add the bridger package to this build to enable the WED offload for 5GHz 
that's mentioned on the same page?

David Lang

On Sun, 4 Feb 2024, Robert James Hernandez wrote:

> Additional info was recently added to the openwrt wiki regarding BSS coloring: https://openwrt.org/toh/linksys/e8450#beam_forming_and_enable_basic_service_set_bss_coloring
>
>

sarcasticadmin, 2024-02-05:
yes its on the same page

davidelang, 2025-02-01:
BSS coloring has the potential to make things worse, it gives APs the authority to ignore exiting transmissions and transmit on top of them.

This can work when you have carefully setup the network and how the channel overlaps should be handled, but we do not have that sort of information. As such I do not think that we can implement this without serious risk of making the problem worse.

@sarcasticadmin

sarcasticadmin, 2025-02-04:
close since I dont think this is going to be something we can support anytime soon

closed 2025-02-04 by sarcasticadmin