socallinuxexpo/scale-network #676: init datasource test for switch-config types

sarcasticadmin, 2024-03-02:
## Description

Currently we dont have any tests that cover the configuration for types: https://github.com/socallinuxexpo/scale-network/tree/8cb8aab0615785177693cfe06b8f0fab51756d30/switch-configuration/config/types

Examples of similar tests would be something like `test_switchtypes_tsv`: https://github.com/socallinuxexpo/scale-network/blob/8cb8aab0615785177693cfe06b8f0fab51756d30/facts/test_datasources.py#L80-L99

## Acceptance Criteria
- verifying the number of fields
- verify the trunktypes (e.g. "Uplink", "Downlink", "AP", etc.)

sarcasticadmin, 2024-12-02:
@owendelong There were a few efforts to attempt to create switch config testing for types https://github.com/socallinuxexpo/scale-network/pull/784 and https://github.com/socallinuxexpo/scale-network/commit/59c4aff211aeb7530cabeae697921f03c1696a72

Neither of these accurately capture how the existing switch-configuration scripts parses switch types. I looks like we'd be better off using the same logic from the perl scripts since its a non standard schema for multiline parsing.

Leaving this note here since there was a fair amount of work around the above attempts. Thank you @cjmakes and @kylerisse for taking a try at this issue :rocket:

owendelong, 2024-12-05:
The multiline parsing is very simple and pretty "unix standard".

A line which matches /\s+\\\s*$/  (a line ending in one or more whitespace characters followed by \ followed by zero or more white space characters) will have the trailing \ removed. The leading spaces will be removed from the next line and then the lines are joined.

owendelong, 2025-02-02:
Happy to write validation code in PERL if you tell me what it needs to output. If someone wants it in Python, someone else gets to write the parser.

owendelong, 2025-02-02:
There's virtually no chance this happens before this year's show.