socallinuxexpo/scale-network #678: raspberry pis need serial number and mac updates in facts

sarcasticadmin, 2024-03-02:
## Description

We now have the raspberry pi 4 hats for POE. I believe that will require use to update pilist.csv https://github.com/socallinuxexpo/scale-network/blob/d695fc5f1121b4715a0b602508e4d0fe600a8d39/facts/pi/pilist.csv

Additionally we they dont have uniq serials like the APs. This would be nice so that we can properly track them as the names of on the devices will change depending on where they are deployed.

## Acceptance Criteria
- pilist.csv updated to include new mac addresses from pi4 POE hats
- serial scheme for pis (should probably include the model)
- serial added to pilist.csv

davidelang, 2024-03-02:
the PoE hats should not change their MAC address

David Lang

On Sat, 2 Mar 2024, Robert James Hernandez wrote:

> ## Description
>
> We now have the raspberry pi 4 hats for POE. I believe that will require use to update pilist.csv [...]

sarcasticadmin, 2024-03-03:
well thats good to hear.

Regardless, we need to set a serial number and label them with their new cases.

closed 2025-11-13 by sarcasticadmin