socallinuxexpo/scale-network #679: caption vm for AV

sarcasticadmin, 2024-03-02:
## Description

A VM that has access from the attendees' WiFi and has incoming access from the AV network.

Agreed upon requirements:

```
2 CPUs
4GB of ram
40GB of disk
Ubuntu 22.04
port 80 (html) open for access to the attendees and some other port open to the A/V network
```

@owendelong is aware of this for 21x

## Acceptance Criteria
- vm deployed to hypervisor1 (hypervisor conference) per requirements
- DNS entry pointing to:  `captions.scale.lan`

owendelong, 2024-03-06:
Please plan on implementing the port limitations via nft or iptables running on the machine (preferably iptables-legacy if I get to choose).

sarcasticadmin, 2024-03-06:
Im leaving the OS configuration up to the AV team to configure

owendelong, 2024-03-07:
That’s fine as long as they are informed that they are responsible for any filtering they want.

Owen

sarcasticadmin, 2024-03-21:
Closing as this was completed and AV was able to use the VM during the expo

closed 2024-03-21 by sarcasticadmin