socallinuxexpo/scale-network #691: pi 'linkrunner' image

davidelang, 2024-03-12:
the linkrunner tool that the facilities folks use seems useful, @MatthewCroughan please see if you can use nix to create a work-alike

this would be a pi image that can run on any pi (possibly limit to pi3 or newer that have built-in wifi)
the pi runs wifi in AP mode
runs a webserver who's index page shows any lldp data and refreshes every few seconds

putting this image on a pi with a battery pack lets us plug into a wire that has link, and it tells us what port on the switch (and what switch) it's connected to. This ends up being easier to find the wire in the closet than using the analog toners

davidelang, 2024-03-14:
also, have the system run DNS, possibly captive portal style that will redirect all browsing traffic to itself. Can it make this be the 'sign in' page for the connection? (I think that's triggered by attempting to go to a specific URL to see if the browser/android/iphone has direct connectivity or is behind a portal, and if behind a portal, triggers the sign-in process.

owendelong, 2024-03-15:
I would think that a GPIO or video-port attached touch-screen (many options available for Pi and other SBCs) would make more sense than a web server.

owendelong, 2024-03-15:
Tests I would want in such a device:
List LLDP Neighbors (names only with drill down on name for full information)
DHCP Address received (v4, v6)
SLAAC, RA, RS information
Ability to capture a small pcap file uploaded to a server we'd prepare for receiving those.
DNS test
PING test
mtr
Display OSPF Hello data

davidelang, 2024-03-15:
For building a dedicated device, I would agree. But the webserver has the 
advantage that it can be an image thrown on any available pi as needed.

David Lang

On Thu, 14 Mar 2024, Owen DeLong wrote:

> I would think that a GPIO or video-port attached touch-screen (many options available for Pi and other SBCs) would make more sense than a web server.