socallinuxexpo/scale-network #746: Signs image - Display

jeffreytjung, 2024-03-17:
## Description
The signs need to have a clock that on the display. The signs show the schedule of talks, and it includes the time that the talk takes place. It is convenient to have a display to show the current time.

## Acceptance Criteria
Editing of the image for the signs Raspberry pi. Placing a clock in the upper center would be best. Connecting to the NTP server on the backend would work.

kylerisse, 2024-03-17:
This actually exists in the codebase already https://github.com/socallinuxexpo/scale-signs/blob/dbdd1d815394821747e1ac6ca9216722da77b1d7/server/js/clock.js#L4
The reason we disabled it years ago is that the code crashes the clients. That being said, it seems like reasonable feature to add back. Will discuss with @kurikurichan !

sarcasticadmin, 2026-02-21:
Signs sever moved to https://github.com/kylerisse/go-signs and the clock was implemented: https://github.com/kylerisse/go-signs/pull/13

MrHamel, 2026-02-22:
Can that repo get moved to the socallinuxexpo umbrella? To at least keep it inline with the kiosk Nix RPi repo.