socallinuxexpo/scale-network #751: Incoporate notification of rate limiting in conference checklist

sarcasticadmin, 2024-03-22:
## Description

Update docs/conference checklist: https://docs.google.com/document/d/1bIdJmbOmKFKKf7YQjrVjy0aw3g8ajMC8krswPXb-GyM/edit?usp=sharing

The goal is to lessen the potential of attendees being rate limited while at the conference.

The following potential rate limit providers that should be contacted prior to the conference starting with our public IPv4 address and IPv6 subnet:

1. GitHub 
2. Docker
3. Fastly (they front cache.nixpkgs.org and seemed to slow us down at one point in both the conference center and hotel)

Additionally it would be a good idea to have user workarounds: Auth Token, etc.

## Acceptance Criteria
- Documentation in conf checklist for providers that could rate limit attendees
- Documentation for workarounds to rate limits imposed by providers

owendelong, 2024-03-22:
We never get the public IPv4 address before Monday.

IPv6 is a non-issue as they don’t seem to rate-limit on the /48, but on the /128.

Since every host gets an individual IPv6 address, there’s a unique /128 per device.

Owen

sarcasticadmin, 2024-03-22:
> We never get the public IPv4 address before Monday.

Yep thats why its in the conference instead of preconf checklist

> IPv6 is a non-issue as they don’t seem to rate-limit on the /48, but on the /128.

Thats good to know, I figured we could cover our basis on ipv6 just so each provider has that information.

genebean, 2024-03-23:
> We never get the public IPv4 address before Monday.

Do you know why this is @owendelong?

sarcasticadmin, 2024-03-23:
> Do you know why this is @owendelong?

PCC and company allocate the ipv4 address off their existing router when we show up on monday.

genebean, 2024-03-24:
Seems odd they wouldn't know that address well in advance as I seriously doubt it changes... maybe I'm over thinking it though

owendelong, 2024-03-26:
It does. They’ve changed ISPs at least 3 times in the time we’ve been holding our event there. Prior to this year, we were getting a /29 from them. This year, we got a single /32.

It appears that they have reduced their overall range from it’s previous /26 to now being just a /29 for the entire PCC.

I won’t defend or even attempt to explain their conduct, but Encore does what Encore does and we just have to cope with it as best we can.

irabinovitch, 2024-03-26:
We reduced to a /32 because we said we didn't need more than 1 IP and it allowed significant savings.

MrHamel, 2024-03-29:
Is the significant savings worth the blowback from the rate limiting? Owen is against running a NAT pool on the firewalls, but it **could** make it **less** of an issue, if for example we had two IPs (one for each building).

@sarcasticadmin I have a few contacts at Fastly that I can poke, to see if they're willing to make some exceptions, can't make any promises though.

irabinovitch, 2024-03-29:
Given we paid many many thousands of dollars per year for 5+ years for the extra IPs and never used them, yes its worth the savings.

davidelang, 2024-03-29:
On Thu, 28 Mar 2024, Ryan Hamel wrote:

> Is the significant savings worth the blowback from the rate limiting? Owen is 
> against running a NAT pool on the firewalls, but it **could** make it **less** 
> of an issue, if for example we had two IPs (one for each building).

I would prefer not to break sessions when people move from one building to the 
other.

David Lang

sarcasticadmin, 2024-03-29:
> Is the significant savings worth the blowback from the rate limiting? Owen is against running a NAT pool on the firewalls, but it could make it less of an issue, if for example we had two IPs (one for each building).

Considering we got rate limits in < 1hr on Thursday I dont see there being much benefit to having a few more IPs since it doesnt by us much additional overhead for the number of requests.

> @sarcasticadmin I have a few contacts at Fastly that I can poke, to see if they're willing to make some exceptions, can't make any promises though.

:+1: thatd be much appreciated

sarcasticadmin, 2024-03-29:
> I would prefer not to break sessions when people move from one building to the
other.

@davidelang the address space between the buildings for wireless is already different: https://github.com/socallinuxexpo/scale-network/blob/c6e657d6cca7fc17aac370905dd1e8105d087f8b/switch-configuration/config/vlans.d/Expo#L2-L3

https://github.com/socallinuxexpo/scale-network/blob/c6e657d6cca7fc17aac370905dd1e8105d087f8b/switch-configuration/config/vlans.d/Conference#L2-L3

davidelang, 2024-03-29:
On Fri, 29 Mar 2024, Robert James Hernandez wrote:

> other.
>
> @davidelang the address space between the buildings for wireless is already different: [...]

yep, realized that just after I sent the message, sorry.

David Lang

owendelong, 2024-03-30:
Actually, we did use them. We made changes this year to reduce our IPv4 utilization and we also had used them previously some years for an Akamai cache server. We don’t need them any more,  but it is not far to say we did not use them. This year is also the first time you mentioned that the multiple IPv4 addresses were costing us extra. Owen

owendelong, 2024-03-30:
I would prefer not to break sessions when people move from one building to the 
other.
They break anyway because their internal address will change. I don’t see a significant gain to doing separate addresses per building because the rate limiting was mostly workshop related which meant it was mostly in one building anyway. 
David Lang