socallinuxexpo/scale-network #754: Reenable kea dhcpv6 on core machines

sarcasticadmin, 2024-03-22:
## Description

Currently kea dhcpv6 daemon is disabled (https://github.com/socallinuxexpo/scale-network/blob/c6e657d6cca7fc17aac370905dd1e8105d087f8b/nix/machines/core/common.nix#L62) due to errors within the dhcpv4 and dhcpv6 daemons clobbering each others pids (these werent originally captured when the issue was happening). I believe this was due to the version of nixpkgs were currently pinned to that happened to include a previous change that was ultimately reverted: https://github.com/NixOS/nixpkgs/pull/274460

## Acceptance Criteria
- Identify that both daemons are no longer errorring with pid race condition

owendelong, 2024-03-25:
The issue isn't just the PIDs.

The issue is the entire /var/run/kea directory.

Each creates its own instance of that directory (some form of kernel namespace systemd thing I admit I don't fully understand and haven't encountered prior to our use of NIX).

Whichever one starts last ends up being the only one visible to users on the system. It is possible there is some way to access the other one which I am unaware of.

Since I was writing debug log files in that directory to overcome systems logging limitations, this was problematic.

owendelong, 2024-03-25:
Yes, the bug description for NixOS 2774460 appears to be exactly what we encountered here.

sarcasticadmin, 2024-03-25:
Thanks for the additional context @owendelong

owendelong, 2024-12-16:
Marking critical as we really want to make sure this is addressed before the show.

sarcasticadmin, 2025-02-06:
Fixed as a result of bumping nixpkgs to d51c28603def282a24fa034bcb007e2bcb5b5dd0 in https://github.com/socallinuxexpo/scale-network/pull/789

Owen and I have confirmed that there is no longer an issue with running `kea-dhcpv4` and `kea-dhcpv6` on the same host.