socallinuxexpo/scale-network #756: Implementing Grafana dashboards for overall network and device health

BerkhanBerkdemir, 2024-03-24:
In SCALE 21x, we discussed about potential improvements in Grafana installation. Currently, we only use OpenWRT dashboard, but it does not highlight overall network status.  So, this issue is a place to keep track each request/idea.

I would like to highlight some of these requests/ideas from the team members.

- [ ] Overall state/health of each networked device such as Raspberry Pis, Access Points (OpenWRT), Switches (Juniper).
  * Total of +200 devices to graph
- [ ] Conference map overlay to visualize status of physically located device
- [ ] Bandwidth usage and health check on Juniper ports
- [ ] Some team members concerned about alarm fatigue.  Alarms will only highlight critical problems
  * tech team members less concerned about Slack and notifications
- [ ] Collect stats from DHCP (Kea) and DNS

This list is not limited, so you are encouraged to add more items in here.

sarcasticadmin, 2024-03-25:
Thanks @BerkhanBerkdemir this looks like a great list to start with

BerkhanBerkdemir, 2024-09-22:
I couldn't attend this weekend to work on this issue. Do we have anyone who wants to work with me on this feature, @sarcasticadmin? I'd like to close this issue with proper PR before December.

fossadmin, 2024-09-23:
Hey @BerkhanBerkdemir, not too familiar with Grafana but willing to learn and help out as much as possible.

sarcasticadmin, 2024-09-24:
@BerkhanBerkdemir thanks for checking back in and no worries about not being able to make the work party.

I think setting a goal of December sounds like a good idea. I think it'd be good to work with @jshcmpbll and @fossadmin on this. I know @jshcmpbll has been working on getting a vmTest together for grafana to do some simple sanity checks on its deployment.

I'd be happy to setup a call to sync up with those who are interested if that sounds like a good idea.

sarcasticadmin, 2026-02-01:
https://github.com/socallinuxexpo/scale-network/pull/983 initial work toward getting a glimpse into overall network and device health.