socallinuxexpo/scale-network #757: Graylog for rsyslog + NetFlow/sFlow Data Collection

MrHamel, 2024-03-29:
## Description
Would it be wise to use something like [Graylog](https://graylog.org/products/source-available/), that can ingest [syslog data, NetFlow/sFlow data, among other things](https://go2docs.graylog.org/5-2/getting_in_log_data/inputs.htm), [make it searchable](https://go2docs.graylog.org/5-2/making_sense_of_your_log_data/how_to_search_your_log_data.html?tocpath=Searching%20Your%20Log%20Data%7C_____0), [and have a dashboard to present data from](https://go2docs.graylog.org/5-2/interacting_with_your_log_data/dashboards.html?tocpath=Aggregating%20Data%7C_____2)?

[What is Graylog?](https://go2docs.graylog.org/5-2/what_is_graylog/what_is_graylog.htm)

[Grafana also supports ElasticSearch as a data source](https://grafana.com/docs/grafana/latest/datasources/#built-in-core-data-sources) if we want to stick to a single dashboard system and solely use Graylog as ingestion, or have two dashboards.

## Acceptance Criteria
Discussion on using Graylog.

davidelang, 2024-03-29:
rsyslog can also send data to elasticsearch.

adding a netflow connector is an interesting idea, but I'd much rather have 
simple syslog to files working again than have a partially implemented fancier 
system that's not quite there.

David Lang

owendelong, 2024-03-30:
I’m hesitant to adopt anything that has a dependency on Mongo.

closed 2026-02-12 by MrHamel