socallinuxexpo/scale-network #760: Update SRXs for recent CVE

owendelong, 2024-07-02:
## Description
https://supportportal.juniper.net/s/article/2024-07-Out-of-Cycle-Security-Bulletin-Junos-OS-SRX-Series-Specific-valid-traffic-leads-to-a-PFE-crash-CVE-2024-21586?language=en_US

## Acceptance Criteria
Once the SRX's have been updated to a release that does not have this vulnerability, it shouldn't be an issue.

owendelong, 2024-12-05:
At this point, we should be looking to update the SRXs to 24.2R1.17 and also make sure that the boot loader is upgraded in the process.

owendelong, 2024-12-05:
I should be able to do this as part of the show prep or at least during the show for 22x.

davidelang, 2024-12-05:
can we plan this for the next work party?

Owen DeLong wrote:

> I should be able to do this as part of the show prep or at least during the show for 22x.

owendelong, 2024-12-13:
If someone has the image and wants to do it, sure. If I can make it to the next work party, fine.
Otherwise, it's really not that big of a deal to do it after arrival for the show.

> can we plan this for the next work party? Owen DeLong wrote:
> […](#)
> I should be able to do this as part of the show prep or at least during the show for 22x.

owendelong, 2024-12-15:
Image is now available from http://dhcp-01.delong.com/images/

owendelong, 2025-02-02:
Ryan has updated all of the SRX software.

closed 2025-02-02 by owendelong