socallinuxexpo/scale-network #761: rpi4 firmware should be set to boot via USB

sarcasticadmin, 2024-09-03:
## Description

Relates to: https://github.com/raspberrypi/firmware/issues/1808

The rpi4 firmware should be set to boot from flash.

Currently, its set to boot from dhcpv6 and then after timeout it will boot

This should be done via the firmware setting the appropriate options

## Acceptance Criteria
- rpi4 boot ignores dhcpv6 until upstream rpi4 fix

davidelang, 2024-09-03:
are we giving up on the Pi firmware supporting netboot?

David Lang

On Tue, 3 Sep 2024, Robert James Hernandez wrote:

> Date: Tue, 03 Sep 2024 13:27:19 -0700
> From: Robert James Hernandez ***@***.***>
> Reply-To: socallinuxexpo/scale-network
>     ***@***.***>
> To: socallinuxexpo/scale-network ***@***.***>
> Cc: Subscribed ***@***.***>
> Subject: [socallinuxexpo/scale-network] rpi4 firmware should be set to boot
>     via USB (Issue #761)
> 
> ## Description
>
> Relates to: https://github.com/raspberrypi/firmware/issues/1808
>
> The rpi4 firmware should be set to boot from flash.
>
> Currently, its set to boot from dhcpv6 and then after timeout it will boot
>
> This should be done via the firmware setting the appropriate options
>
> ## Acceptance Criteria
> - rpi4 boot ignores dhcpv6 until upstream rpi4 fix
>
>
>

sarcasticadmin, 2024-09-03:
no but its currently a firmware limitation with the pi

davidelang, 2024-09-03:
I thought that the status as of last year was that the Pi could boot from a 
dhcp6 server with everything on the same network, but could not work with one on 
a different subnet.

David Lang

On Tue, 3 Sep 2024, Robert James Hernandez wrote:

> Date: Tue, 03 Sep 2024 13:38:40 -0700
> From: Robert James Hernandez ***@***.***>
> Reply-To: socallinuxexpo/scale-network
>     ***@***.***>
> To: socallinuxexpo/scale-network ***@***.***>
> Cc: David Lang ***@***.***>, Comment ***@***.***>
> Subject: Re: [socallinuxexpo/scale-network] rpi4 firmware should be set to
>     boot via USB (Issue #761)
> 
> no but its currently a firmware limitation with the pi
>
>

owendelong, 2025-02-02:
The status as of last year was mostly confusion.

The status as of now is that if the tftp and DHCP servers are one and the same machine and both are on-link with the pi, it might be possible (further testing is needed).

But the PIs should NEVER (IMHO) be set to boot from USB. They should only be set to boot from SD card and NETBOOt. IMHO, we should swap the order to prefer SD and then IPv6 so that if there's an SD card present, it will boot from that and if not it will try IPv6. We should prefix the boot order variable with 0xf so that it will retry the entire list if the left most choice fails (the list is parsed right to left).

https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER has a list of possible values.

Based on this list, my current recommendation would be 0xf21 as the SD card will fail very rapidly if not present, so the trivial wait for netboot won't matter, but the extensive wait for netboot won't pre-empt SD booting.

owendelong, 2025-02-22:
Further research revealed that even on the same network, the Pi just plain doesn't do sane things with IPv6 netbook. The powers that be don't intend to work on it, so we're basically stuck until they change their mind.

At this point, I'm going to be looking for alternative platforms that can run off of eMMC or Netboot over IPv6.

closed 2025-02-22 by owendelong