socallinuxexpo/scale-network #763: test 3d prints for drive trays

sarcasticadmin, 2024-09-21:
## Description

Right now the IBM servers done have trays for some of drives

[IBM system x3650 tray](https://www.printables.com/model/263072-ibm-system-x3650-m4-25-tray)

## Acceptance Criteria
- Test print and fitment in x3650 server

ptrlow1, 2024-09-22:
[image]

[image]

Printed and tested to be working. (See the attached photos)

Would need to buy these screws to secure the drives to the 3D printed caddy:
uxcell 50pcs Hard Drive Screws M3 x 4.7mm HDD SSD Hard Drive Mounting Screws Black Metal Hard Drive Replacement Screws for Laptop Hard Drive 
> https://a.co/d/6cpOk39

ptrlow1, 2024-09-22:
Installed the new trays in the 2 servers.
Used the screws that secure the SSDs in the caddy that were in screw bins @ Null Space Labs.

[image]

Recommend to close this as completed.

sarcasticadmin, 2024-09-22:
thank you very much @ptrlow1 the prints came out great!

closed 2024-09-22 by sarcasticadmin