socallinuxexpo/scale-network #772: resolution of pi4 signs and reg

sarcasticadmin, 2024-09-22:
## Description

Lets confirm that the scaling for signs and reg is appropriate given that we have a 4k and 1080p mix of displays

## Acceptance Criteria

- Reg is an acceptable resolution
- Signs is an acceptable resolution

ruebenramirez, 2024-11-03:
I've been dabbling with kanshi on sway.  I'm interested in digging into this problem, as long as I'm not stepping on anyone else's toes :)

sarcasticadmin, 2024-11-03:
> I've been dabbling with kanshi on sway. I'm interested in digging into this problem, as long as I'm not stepping on anyone else's toes :)

absolutely, we'd appreciate the help. @MatthewCroughan has been leading the charge on the pi image building and has some good context from last year and the challenges.

MatthewCroughan, 2024-11-03:
Kanshi is interesting and I've wanted to use that myself. I hadn't considered it as a solution to this problem, but I suppose it could be! I can try this myself also.

ruebenramirez, 2024-11-04:
I did a little write up of my experience with Kanshi here: https://blog.rueb.dev/posts/2024-10-01-display-configuration-with-kanshi/

I'll dig into the pi image this week to see how it's all put together.  I'm looking forward to learning how everything works.

MatthewCroughan, 2024-11-04:
@ruebenramirez Thank you for taking an interest, I'm sure I'll get to shake your hand at SCALE/PlanetNix, please ping me if you need anything explained/disambiguated.

ruebenramirez, 2024-11-27:
Hey guys!  I have a WIP PR with the general idea of kanshi managing 4k and 1080p displays here: https://github.com/socallinuxexpo/scale-kiosk/pull/8

I still consider myself something of a novice with NixOS, so I'm sure there will be plenty to update in this PR to get it into proper shape.  :)

ruebenramirez, 2024-11-30:
@MatthewCroughan was able to get this over the finish line with his awesome work here: https://github.com/socallinuxexpo/scale-kiosk/pull/13.  I think this is ready to close now.

closed 2024-11-30 by ruebenramirez