socallinuxexpo/scale-network #773: output switch-maps pdf as part of the release

sarcasticadmin, 2024-09-22:
## Description

Switch-maps should be included as a release artifact. They are simple enough to generate: `make config/switch-maps/stickers.pdf`

## Acceptance Criteria

- switch-maps are an artifact of the release

owendelong, 2024-12-05:
Can you clarify what needs to be done for this to be achieved? What is a "release" in this context and forgive my ignorance as to the definition of the term artifact in this context.

I suspect that if we're including the sticker PDF in the release, we should probably be including the full set of switch configurations, etc. as well, in which case, "make all" or "make" without arguments is probably the better choice for the generation process.

sarcasticadmin, 2024-12-09:
> Can you clarify what needs to be done for this to be achieved? What is a "release" in this context and forgive my ignorance as to the definition of the term artifact in this context.

A release is a concept of associating additional files with things like a `tag`. For example see one of our recent releases where we include the openwrt images with the `21.0` tag: https://github.com/socallinuxexpo/scale-network/releases/21.0

> I suspect that if we're including the sticker PDF in the release, we should probably be including the full set of switch configurations, etc. as well, in which case, "make all" or "make" without arguments is probably the better choice for the generation process.

We need to include it in a github action that creates a release when a given git tag is created.

owendelong, 2025-02-02:
Well... Basically, if you run a "make" from within switch-configuration, you'll get all the artifacts in switch-configuration/config/{output, switch-maps}.

You probably care about ...output/* and ...switch-maps/*.pdf