socallinuxexpo/scale-network #775: reallocate switches

davidelang, 2024-09-22:
organize the switches in numberic order in the carts

switches 1-11 noc cart
switches 12-23 lower conf cart
switches 24-39 expo cart
switches 40-51 ballroom cart
switch 52 monitor cart

new order is:

1 NOC
2 ConfIDF
3 NW-IDF
4 NE-IDF
5 ExpoIDF
6 Expo-Catwalk
7 MassFlash
8 AVSwitch
9 Rm209-210
10 Rm211
11 Rm214
12 Rm101-102
13 Rm103
14 Rm104
15 Rm105
16 Rm106
17 Rm107
18 SpareA
19 SpareB
20 SpareC
21 SpareD
22 SpareE
23 SpareF
24 ExpoAW
25 ExpoA1
26 ExpoA2
27 ExpoA3
28 ExpoA4
29 ExpoA5
30 ExpoB1
31 ExpoB2
32 ExpoB3
33 ExpoB4
34 ExpoB5
35 ExpoC1
36 ExpoC2
37 ExpoC3
38 ExpoC4
39 ExpoC5
40 BallroomA
41 BallroomB
42 BallroomC
43 BallroomDE
44 BallroomF
45 BallroomG
46 BallroomH
47 SpareG
48 SpareH
49 SpareI
50 SpareJ
51 SpareK
52 RegDesk

davidelang, 2024-09-22:
@kylerisse  issue for the work you are doing.

closed 2024-09-22 by owendelong

kylerisse, 2024-09-23:
https://gist.github.com/kylerisse/61cde5f50950b16f0901c2cf07429864

PR #779 (2024-09-22): [READY] Reallocate switches
Fixed #775 

## Description of PR
This is a mass renaming of all the switches per work party 1. It also disables the data validation for switch noise level as that will be going away.

## Previous Behavior

- switches had previous IDs
- noise level data validation only allowed for specific values

## New Behavior

- switches have new IDs supplied by @davidelang 
- noise level data validation now passes with any value, no longer strict

## Tests

1. spot checked a couple of entries in the output
2. ran `nix build -L .#checks.x86_64-linux.pytest-facts` to validate TSV [... cut]