socallinuxexpo/scale-network #776: Wi-Fi MAC Address Count Monitoring & Alerting

MrHamel, 2024-09-22:
## Description

Document MAC address limitations of routers switch chips, which is important for usage monitoring and exhaustion alerting.

Belkin RT3200/Linksys E8450 - 2K MACs
Netgear WNDR3800 - 1K MACs

## Acceptance Criteria

Grafana monitoring and alerting of the 1K client threshold (lowest common denominator) in either building.

davidelang, 2024-09-22:
see the option
option 'enable_vlan4k' '1'
to enable vlan ids up to 4k on the 3800s

David Lang

On Sun, 22 Sep 2024, Ryan Hamel wrote:

> Date: Sun, 22 Sep 2024 14:35:42 -0700
> From: Ryan Hamel ***@***.***>
> Reply-To: socallinuxexpo/scale-network
>     ***@***.***>
> To: socallinuxexpo/scale-network ***@***.***>
> Cc: Subscribed ***@***.***>
> Subject: [socallinuxexpo/scale-network] Wi-Fi MAC Address Count Monitoring &
>     Alerting (Issue #776)
> 
> ## Description
>
> Document MAC address limitations of routers switch chips, which is important for usage monitoring and exhaustion alerting.
>
> Belkin RT3200/Linksys E8450 - 2K MACs
> Netgear WNDR3800 - 1K MACs
>
> ## Acceptance Criteria
>
> Grafana monitoring and alerting of the 1K client threshold (lowest common denominator) in either building.
>
>

MrHamel, 2024-09-22:
Please provide the documentation where it allocates more resources in hardware on the switch chip. Thanks!

davidelang, 2024-09-22:
I misunderstood the issue and thought this was about the vlan ids, instead of 
the total mac addresses that the chip will keep track of.

As we are currently using the 3800s, this is not a problem because we don't have 
the wifi bridge VLANs going through the switch, they go over the uplink port 
which is connected directly to the CPU.

On the new APs, there is no option to bypass the switch.

The question is what happens when the mac table is overloaded.

1. does the switch go into open mode, forwarding the packet to all ports?
or
2. does the switch dump some other MAC from it's table, possibly causing 
thrashing of the mac table (and a lot of arp traffic)
or
3. does the switch crash or otherwise become unavailable.

One work-around for this is to create a tunnel between the CPU on the APs and 
the router (so that the switch chips don't see the wifi mac addresses). In a 
discussion, Ryan mentioned a 'standard' way to tunnel layer 2, but mentioned 
that juniper devices as old as ours don't support it

the other thing would be to implement mobile IP addresses and route the wifi 
clients at the AP level, I believe this is still 'lab research' level of 
implementation.

sarcasticadmin, 2025-02-04:
@MrHamel can you do this for the current hardware we are using and put it into markdown?

I would include `Openwrt One` in the list as well.

MrHamel, 2025-02-09:
- Belkin RT3200/Linksys E8450 - 2K MACs (https://repo.librerouter.org/misc/lr2/MT7531_switch_Reference_Manual_for_Development_Board.pdf)

- Netgear WNDR3800 - 1K MACs (http://realtek.info/pdf/rtl8366s_8366sr_datasheet_vpre-1.4_20071022.pdf)

- OpenWRT One - No limit on the SoC or Wi-Fi chipset.

davidelang, 2025-02-09:
the openwrt one does not have a switch chip, just two network ports to the CPU

But I thought the concern was MAC limits on the wifi chips, not the switch 
chips.

David Lang

On Sun, 9 Feb 2025, Ryan Hamel wrote:

> Date: Sun, 09 Feb 2025 09:21:06 -0800
> From: Ryan Hamel ***@***.***>
> Reply-To: socallinuxexpo/scale-network
>     ***@***.***>
> To: socallinuxexpo/scale-network ***@***.***>
> Cc: David Lang ***@***.***>, Comment ***@***.***>
> Subject: Re: [socallinuxexpo/scale-network] Wi-Fi MAC Address Count Monitoring
>      & Alerting (Issue #776)
> 
> - Belkin RT3200/Linksys E8450 - 2K MACs (https://repo.librerouter.org/misc/lr2/MT7531_switch_Reference_Manual_for_Development_Board.pdf)
>
> - Netgear WNDR3800 - 1K MACs (http://realtek.info/pdf/rtl8366s_8366sr_datasheet_vpre-1.4_20071022.pdf)
>
> - OpenWRT One - No limit on the SoC or Wi-Fi chipset.
>
>

MrHamel, 2025-02-09:
> Caution: This is an external email and may be malicious. Please take care when clicking links or opening attachments.

The issue lies within the switch chip, anything beyond that is all Linux + RAM. It makes more sense to switch to the OpenWRT One, than to implement a crazy convoluted VXLAN hack to keep old hardware working.

Ryan Hamel

owendelong, 2025-02-10:
> see the option
> option 'enable_vlan4k' '1'
> to enable vlan ids up to 4k on the 3800s
> 
> David Lang
> […](#)

I don't think that's related to MAC address limits, and we have no plans that I'm aware of to go use any VLAN IDs beyond 1024.

owendelong, 2025-02-10:
For once, I agree with @MrHamel 

Documenting the limits and tracking them in Grafana makes sense.

I don't think we're likely to be hitting them, given the Attendee/Access point ratio.

Unless more than 1/8 of the attendees for the entire show (across its full duration, without regard to people coming and going at different times) all hit a single access point at once, we shouldn't have an issue even with the 1K MAC limit on the 3800s.

Is that impossible? No, but highly improbably. Sure, we don't get perfect distribution of X attendees across Y access points, but still, with 80+ access points deployed, if we have somehow get X/8 attendees hitting a single access point, we've got some larger problem than the AP running out of L2 forwarding table space.

As to what happens when they run out, that's behavior that is implementation dependent and usually not documented in public. We could conduct experiments to find out, of course, but that's a significant amount of effort and I think hardly worth it given the low probability of this being an actual problem.

MrHamel, 2025-02-10:
From http://realtek.info/pdf/rtl8366s_8366sr_datasheet_vpre-1.4_20071022.pdf

  *
8.4. Search and Learning
  *
The RTL8366S/SR supports 8-entry Content Address Memory (CAM) to avoid look-up table hash collision. When all 4-entry in the look-up table index are occupied, the source MAC address can be learned into one of the 8-entry CAM. If both look-up table and CAM are full, the source MAC address will not be learned in the RTL8366S/SR.

Based on the datasheet, it's stops learning when it's full. Now I don't believe it would fall over, it'll start broadcasting.

owendelong, 2025-02-10:
In which case, I don't see it as a serious problem since we only have three real interfaces on the device anyway (2 RF and ethernet) and while splattering the 2.4. band with copies of the 5G traffic is annoying, I consider it more of a "meh" if it ever happens than a worry.

Everything that comes in on the radios likely needs to go out the ethernet port anyway, so that won't be a significant issue.

davidelang, 2025-02-10:
Ryan Hamel wrote:

> From http://realtek.info/pdf/rtl8366s_8366sr_datasheet_vpre-1.4_20071022.pdf
>
>  *
> 8.4. Search and Learning
>  *
> The RTL8366S/SR supports 8-entry Content Address Memory (CAM) to avoid look-up table hash [...]

correct, and a hash is not going to be perfectly distributed.

But this is talking about the 5 port switch chip in the wndr3800s

one port goes to the CPU, the other 4 ports are exposed on the back of the 
router.

This says nothing about the wifi. The wifi is not connected to this chip at all.

In the last few years, we have only gone through this chip for programming. We 
have used the WAN port (which also doesn't go through this chip) for our uplink 
and the wifi network has been a VLAN on this trunked port.

So this is not relevant to concerns about connection limits to the wifi 
chipsets.

David Lang

MrHamel, 2025-02-11:
That is correct, making the Belkin/Linksys devices the only ones we have to worry about (although to a lesser degree).

davidelang, 2025-02-11:
David Lang wrote:

> pretty much every ethernet switch has some MAC limit, and the design failure 
> mode is to start broadcasting out all ports. If we run into that problem, we 
> should see bandwith utilization spike on the switch uplink ports that stops 
> when it hits the routers.

What is the limit on the Juniper routers? The switches on the linksys/belkin 
APs are only going to see the MAC addresses that are either directly connected 
to them, or that the juniper uplink sends to them.

David Lang

MrHamel, 2025-02-11:
EX4200's - 16K
EX4300's (what we're eventually going to) - 64K

SRX300 - It's "L2 Accelerated" a.k.a no ASIC, just special sauce built into the Cavium Octeon CPU/SoC.