socallinuxexpo/scale-network #781: substituter errors in CI (GH rate limit)

kylerisse, 2024-09-23:
## Description
Since  #770 , but maybe before, CI showing errors when using the magic cache substituter.

Example from https://github.com/socallinuxexpo/scale-network/actions/runs/10977769534/job/30480116742?pr=770 which is not limited to the new NixOS configuration validations, also impacts the existing checks.

```
copying path '/nix/store/qkpd5prdrc6dnqy0wpcqznr83hzn07pi-freetype-2.13.1' from 'http://127.0.0.1:37515/'...
copying path '/nix/store/bhmynyjwzc2r6iqf7fhc3yzjcv3paiwa-curl-8.3.0' from 'http://127.0.0.1:37515/'...
warning: error: unable to download 'http://127.0.0.1:37515/nar/1hdp17l0h950ym38nsjxbp4m1kvpa7jsdv0l1521q3v7mampmd21.nar.xz': HTTP error 429

       response body:

       GitHub API error: API error (429 Too Many Requests): StructuredApiError { message: "Request was blocked due to exceeding usage of resource 'Count' in namespace ''." }; retrying in 310 ms
copying path '/nix/store/4g9zd3czh9k1vy53f4mhadjqsm5ma82a-boost-1.81.0' from 'http://127.0.0.1:37515/'...
disabling binary cache 'http://127.0.0.1:37515/' for 60 seconds
error: unable to download 'http://127.0.0.1:37515/nar/109hhxp4qqdw32rp6v5ljjk7l3a9k9i9796gdmmi2swhpa23r23y.nar.xz': HTTP error 418

       response body:

       GitHub API error: GitHub Actions Cache throttled Magic Nix Cache. Not trying to use it again on this run.
copying path '/nix/store/1hpgya88k6424flhwrwfli09rq3yxb8q-nmap-7.94' from 'http://127.0.0.1:37515/'...
error: substituter 'http://127.0.0.1:37515/' is disabled
copying path '/nix/store/fmh3s032bcsbfcdp82zsjlmkj1kp72j6-sqlite-3.43.1' from 'http://127.0.0.1:37515/'...
copying path '/nix/store/4g9zd3czh9k1vy53f4mhadjqsm5ma82a-boost-1.81.0' from 'https://cache.nixos.org/'...
copying path '/nix/store/74y3751gsixaz9797ib0hp7c658sp1y5-binutils-2.40' from 'http://127.0.0.1:37515/'...
error: substituter 'http://127.0.0.1:37515/' is disabled
copying path '/nix/store/1hpgya88k6424flhwrwfli09rq3yxb8q-nmap-7.94' from 'https://cache.nixos.org/'...
copying path '/nix/store/agpc9pgdql4s53jzr46gyx6d8qdivppq-strace-6.5' from 'http://127.0.0.1:37515/'...
error: substituter 'http://127.0.0.1:37515/' is disabled
error: substituter 'http://127.0.0.1:37515/' is disabled
copying path '/nix/store/ghf532yx8lvrysj502jvrzqm5987a95j-bind-9.18.19-lib' from 'http://127.0.0.1:37515/'...
```

## Acceptance Criteria

- CI no longer errors when trying to use `magic-nix-cache`

sarcasticadmin, 2024-09-23:
I dont know if theres much we can do about the cache limit: https://github.com/DeterminateSystems/magic-nix-cache-action?tab=readme-ov-file#usage-notes

But we might just want to limit the builds for the machines to manual triggers and to `master`. We could also consolidate which machines we build until we run them on the `devServer` via a hosted runner.

kylerisse, 2024-09-24:
> I dont know if theres much we can do about the cache limit: https://github.com/DeterminateSystems/magic-nix-cache-action?tab=readme-ov-file#usage-notes
>
> But we might just want to limit the builds [...]

Understood, suggest we close this and live with it until it's on hosted runner. Seem good?

djacu, 2024-09-24:
Thoughts on using cachix instead of the magic cache? It would require a secret be stored in the repo and there are security concerns outline [here](https://github.com/cachix/cachix-action?tab=readme-ov-file#security) but it has worked well for me in the past.

sarcasticadmin, 2024-09-24:
> Understood, suggest we close this and live with it until it's on hosted runner. Seem good?

Lets keep it up since we should split up the job and leave the matrixing to run only on `master` and `manual` triggers. This way we dont bump up against the limit for any in coming PR.

> Thoughts on using cachix instead of the magic cache? It would require a secret be stored in the repo and there are security concerns outline [here](https://github.com/cachix/cachix-action?tab=readme-ov-file#security) but it has worked well for me in the past.

the only downside I see here is that we would have to reach out to cachix for each run vs the magic cachix being closer to github actions. I think splitting up the jobs as mentioned above should be sufficient until we get the devServer going, the hosting a cache and runners on it removes many of these headaches 🙂

kylerisse, 2024-09-24:
Given the failures are gracefully handled and fall through to cache.nixos.org anyways, this might be more of a case of *old man yelling at clouds* (me) and sorry logs, than a real problem.

sarcasticadmin, 2024-09-24:
Well I'd like to keep CI as snappy as possible so Im fine limiting shifting things around and well see how it goes

PR #783 (2024-09-27): [READY] - gha: limit nixos builds to master and manual
## Description

Fixes: #781

Running these on every PR or every push to a PR quickly gets us rate limited by github. Its unclear if theres a way around this cache rate limit:
https://github.com/DeterminateSystems/magic-nix-cache-action?tab=readme-ov-file#usage-notes

Moving this to only running on `master` and manually should do the trick for now. If it persists we might have to move it to a scheduled job like openwrt builds.

## Previous Behavior

- Bumping up against gha cache limits for many PRs due to nixosConfiguration builds

## New  [... cut]
closed 2024-09-27 by sarcasticadmin