socallinuxexpo/scale-network #782: Better accommodation for IoT classes using Install VLAN

owendelong, 2024-09-23:
## Description

Currently, we have applied a workaround for IoT (and other trainings) which require wired devices on the same broadcast domain as wireless devices by dedicating 8 ports on each of the cfRoom switches as untagged cfSCALE-FAST.

This is a security risk as well as an improper use of this global WiFi VLAN.

I propose instead, that we add the cfInstall VLAN as an additional "SCALE-TRAINING" SSID on the 2.4Ghz radios only.

This would give us the ability to impose additional limitations on that VLAN (e.g. not allowing it to communicate with other conference VLANs, limitations on its internet access, etc.

I'd also like to see us use semi-smart tabletop switches for all of the training labs that would use these ports. In this way, the ports could be tagged on the cfRoom switches and converted to untagged by the tabletop switches. This would prevent casual use of the ports in non-training rooms by conference attendees.

These are a couple of possible solutions. I'm open to additional suggestions or better ideas.

## Acceptance Criteria

No further untagged ports open on switches in public accessible areas
Better security audit and control of the Install VLAN
Implementation of the above across all cfRoom switches.

davidelang, 2024-09-23:
Owen DeLong wrote:

> ## Description
>
> Currently, we have applied a workaround for IoT (and other trainings) which require wired devices on the same broadcast domain as wireless devices by dedicating 8 ports on each of [...]

ideally this would only be in the room that's doing the training. (first use 
case for our netmap parameter? although I think it only does wired currently)

> This would give us the ability to impose additional limitations on that VLAN (e.g. not allowing it to communicate with other conference VLANs, limitations on its internet access, etc.

yes, we need to do more of this sort of thing

> I'd also like to see us use semi-smart tabletop switches for all of the
> training labs that would use these ports. In this way, the ports could be
> tagged on the cfRoom switches and converted to [...]

sounds good, there are a lot of pretty cheap 8 and 16 port switches out there 
that are low power and small (much better for the tables than using enterprise 
grade switches, even if the price delta isn't that much)

David Lang

> These are a couple of possible solutions. I'm open to additional suggestions or better ideas.
>
> ## Acceptance Criteria
>
> No further untagged ports open on switches in public accessible areas
> Better [...]

owendelong, 2024-09-23:
You'll need to explain the "net map" parameter to me, first I'm hearing of it.

Cheap 8 and 16 port unmanaged switches wouldn't quite cut it in this role. We need at least something that understands VLANs and 802.1q.

owendelong, 2024-09-23:
In other words, something like these:
https://www.amazon.com/TP-LINK-TL-SG1016DE-16-Port-Gigabit-Switch/dp/B00K4DS67C/ref=sr_1_3?crid=2KVNVTGX3YMBK&dib=eyJ2IjoiMSJ9.9sM9a8Hl1mrq8klDWQK4unL6g1l41_hfcri6d85rK0w5GCR3X2cBjluAMbOMScQQZgVTFhXz6E6EQ4UMg8VxyPvcUPCpWT2MyUUR3ePabWS6nAdeuyIU_qLhMgTmq7p8eC3QBsjZv7yiLNio2oJgp1bapNlFsnBEKYiCdoq6oEUiOdgCkbUWDaxc6VhHys2zcvk5ZHwWn3MsL5tPnBLdzX7rDj6tW1NLqvTqJeQ9-zM.CMgOh8vU91LItR7MIRBO167soAkmLnpDeU00R8boKSc&dib_tag=se&keywords=16%2Bport%2Bvlan%2Bswitch&qid=1727128179&sprefix=16%2Bport%2Bvlan%2Bswitch%2Caps%2C149&sr=8-3&th=1

https://www.amazon.com/TP-LINK-Gigabit-Ethernet-Network-TL-SG116E/dp/B07GRG63P6/ref=sr_1_10?crid=2KVNVTGX3YMBK&dib=eyJ2IjoiMSJ9.9sM9a8Hl1mrq8klDWQK4unL6g1l41_hfcri6d85rK0w5GCR3X2cBjluAMbOMScQQZgVTFhXz6E6EQ4UMg8VxyPvcUPCpWT2MyUUR3ePabWS6nAdeuyIU_qLhMgTmq7p8eC3QBsjZv7yiLNio2oJgp1bapNlFsnBEKYiCdoq6oEUiOdgCkbUWDaxc6VhHys2zcvk5ZHwWn3MsL5tPnBLdzX7rDj6tW1NLqvTqJeQ9-zM.CMgOh8vU91LItR7MIRBO167soAkmLnpDeU00R8boKSc&dib_tag=se&keywords=16%2Bport%2Bvlan%2Bswitch&qid=1727128179&sprefix=16%2Bport%2Bvlan%2Bswitch%2Caps%2C149&sr=8-10&th=1

https://www.amazon.com/16-Port-Gigabit-Wallmount-Rackmount-GS1900-16/dp/B0BM7X7CHX/ref=sr_1_1_sspa?crid=1FR75LFWP0AC5&dib=eyJ2IjoiMSJ9.s8A9GIIlTdzvlq-xrnzobe0Ej98aWbxKaIlQ_VBJLcneeTKeLYaYyrXMA7XA1U6TlTSttS2KqM3F-2f19qIpJY7PqSuieOzvMoAWO5kGsyxF3DYI5mrKvSvnzN3QrOwvPmXkJS7cIUBw-pzM0tNK0X-sKaAgdEX0AwvsZ9e0kdUTKFomC21eDkdjqTN4v2YXA6rRB9NYO20THzQ__5nhRlQAxT9CDR_4diZqlIHPOIaZ5E9zfJTzrWiE7ye7C2K_dOqyyk8guBSH1-u5GD7IFnDjX6OF_P27qFWGi0JItH8.aJZWyp6yLOf1GGe-GjjxzfJyID7U-uoSATY_v7L5niA&dib_tag=se&keywords=16+port+managed+switch&qid=1727128401&s=electronics&sprefix=16+port+managed+switch%2Celectronics%2C147&sr=1-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1

https://www.amazon.com/D-Link-Ethernet-Internet-Mountable-DGS-1100-16V2/dp/B0876G7X18/ref=sr_1_13?crid=1FR75LFWP0AC5&dib=eyJ2IjoiMSJ9.s8A9GIIlTdzvlq-xrnzobe0Ej98aWbxKaIlQ_VBJLcneeTKeLYaYyrXMA7XA1U6TlTSttS2KqM3F-2f19qIpJY7PqSuieOzvMoAWO5kGsyxF3DYI5mrKvSvnzN3QrOwvPmXkJS7cIUBw-pzM0tNK0X-sKaAgdEX0AwvsZ9e0kdUTKFomC21eDkdjqTN4v2YXA6rRB9NYO20THzQ__5nhRlQAxT9CDR_4diZqlIHPOIaZ5E9zfJTzrWiE7ye7C2K_dOqyyk8guBSH1-u5GD7IFnDjX6OF_P27qFWGi0JItH8.aJZWyp6yLOf1GGe-GjjxzfJyID7U-uoSATY_v7L5niA&dib_tag=se&keywords=16%2Bport%2Bmanaged%2Bswitch&qid=1727128401&s=electronics&sprefix=16%2Bport%2Bmanaged%2Bswitch%2Celectronics%2C147&sr=1-13&th=1

All in the ~$70-110 price range.

davidelang, 2024-09-23:
Owen DeLong wrote:

> You'll need to explain the "net map" parameter to me, first I'm hearing of it.

we pass a dhcp parameter to the APs that lets them reconfigure their network 
config amoung one of several pre-installed versions. (load multiple network 
configs with basename.# and the # that is passed via dhcp gets copied on top of 
the real file and the AP reboots)

not as flexible as feeding the entire config, but a little more error resistant 
as all options can be pre-tested, and there is a default if there isn't one 
available.

I think it only does the wired config right now, but it could be tweaked 
slightly to cover wireless as well.

> Cheap 8 and 16 port unmanaged switches wouldn't quite cut it in this role. We need at least something that understands VLANs and 802.1q.

this isn't as cheap as some (~$50), but that's because it includes PoE 
https://www.amazon.com/gp/product/B07PY93BL2/

some cheap ones that I haven't used (as cheap as $13 for a 5 port one)
https://www.amazon.com/STEAMEMO-Easy-Managed-Ethernet-Splitter/dp/B0DG2N8DWH
https://www.amazon.com/Gigabit-Managed-Snooping-Aggregation-GS1200-5/dp/B07BNVTZ3S

These are basically using the same switch chips that are used in APs, but 
managed with their proprietary web GUI.

David Lang

owendelong, 2024-09-23:
Since a 5 port switch is effectively a 4 port (plus uplink) switch, the $/port is actually about the same as the 16 port switches I was proposing in the $70 range. $70/15+ usable = $4.67/port while $13/4 usable = $3.25/port. Further, I proposed switches from well established known brands vs. random unknowns (The only recognized brand in your list is the $50 TP-Link which is $7.05/port so more expensive than the options I proposed).

Further, I think managing that many 5 port switches plus their attendant wall warts, etc. would be a much bigger pain than a smaller number of 16 port switches.

I'm willing to go with the group thinking on this, but I think the above considerations are worth factoring.

davidelang, 2024-09-23:
same type of things, there is a big price jump from 8 to 16 ports

$26 for 8 ports
https://www.amazon.com/TP-LINK-Gigabit-Ethernet-Network-TL-SG116E/dp/B00K4DS5KU?th=1

$72 for 16 ports
https://www.amazon.com/TP-LINK-Gigabit-Ethernet-Network-TL-SG116E/dp/B07GRG63P6?th=1

8 ports seems the sweet spot in price (among 5-8-16) and seems a reasonable 
number to put on a desk

This one is PoE powered $13 for 5 ports, $16 for 8 ports, $90 for 16 ports
https://www.amazon.com/STEAMEMO-Easy-Managed-Ethernet-Splitter/dp/B0DGKN4L2F?th=1

I think it would be good to replace our current dumb switches with something 
like any of these 8 port ones (either the Steamemo PoE powered ones or the 
tp-link externally powered ones), but the budget priority would be behind APs or 
the room switches

David Lang

On Mon, 23 Sep 2024, Owen DeLong wrote:

> Date: Mon, 23 Sep 2024 14:55:57 -0700
> From: Owen DeLong ***@***.***>
> Reply-To: socallinuxexpo/scale-network
>     ***@***.***>
> To: socallinuxexpo/scale-network ***@***.***>
> Cc: David Lang ***@***.***>, Comment ***@***.***>
> Subject: Re: [socallinuxexpo/scale-network] Better accommodation for IoT
>     classes using Install VLAN (Issue #782)
> 
> In other words, something like these:
> https://www.amazon.com/TP-LINK-TL-SG1016DE-16-Port-Gigabit-Switch/dp/B00K4DS67C/ref=sr_1_3?crid=2KVNVTGX3YMBK&dib=eyJ2IjoiMSJ9.9sM9a8Hl1mrq8klDWQK4unL6g1l41_hfcri6d85rK0w5GCR3X2cBjluAMbOMScQQZgVTFhXz6E6EQ4UMg8VxyPvcUPCpWT2MyUUR3ePabWS6nAdeuyIU_qLhMgTmq7p8eC3QBsjZv7yiLNio2oJgp1bapNlFsnBEKYiCdoq6oEUiOdgCkbUWDaxc6VhHys2zcvk5ZHwWn3MsL5tPnBLdzX7rDj6tW1NLqvTqJeQ9-zM.CMgOh8vU91LItR7MIRBO167soAkmLnpDeU00R8boKSc&dib_tag=se&keywords=16%2Bport%2Bvlan%2Bswitch&qid=1727128179&sprefix=16%2Bport%2Bvlan%2Bswitch%2Caps%2C149&sr=8-3&th=1
>
> https://www.amazon.com/TP-LINK-Gigabit-Ethernet-Network-TL-SG116E/dp/B07GRG63P6/ref=sr_1_10?crid=2KVNVTGX3YMBK&dib=eyJ2IjoiMSJ9.9sM9a8Hl1mrq8klDWQK4unL6g1l41_hfcri6d85rK0w5GCR3X2cBjluAMbOMScQQZgVTFhXz6E6EQ4UMg8VxyPvcUPCpWT2MyUUR3ePabWS6nAdeuyIU_qLhMgTmq7p8eC3QBsjZv7yiLNio2oJgp1bapNlFsnBEKYiCdoq6oEUiOdgCkbUWDaxc6VhHys2zcvk5ZHwWn3MsL5tPnBLdzX7rDj6tW1NLqvTqJeQ9-zM.CMgOh8vU91LItR7MIRBO167soAkmLnpDeU00R8boKSc&dib_tag=se&keywords=16%2Bport%2Bvlan%2Bswitch&qid=1727128179&sprefix=16%2Bport%2Bvlan%2Bswitch%2Caps%2C149&sr=8-10&th=1
>
> https://www.amazon.com/16-Port-Gigabit-Wallmount-Rackmount-GS1900-16/dp/B0BM7X7CHX/ref=sr_1_1_sspa?crid=1FR75LFWP0AC5&dib=eyJ2IjoiMSJ9.s8A9GIIlTdzvlq-xrnzobe0Ej98aWbxKaIlQ_VBJLcneeTKeLYaYyrXMA7XA1U6TlTSttS2KqM3F-2f19qIpJY7PqSuieOzvMoAWO5kGsyxF3DYI5mrKvSvnzN3QrOwvPmXkJS7cIUBw-pzM0tNK0X-sKaAgdEX0AwvsZ9e0kdUTKFomC21eDkdjqTN4v2YXA6rRB9NYO20THzQ__5nhRlQAxT9CDR_4diZqlIHPOIaZ5E9zfJTzrWiE7ye7C2K_dOqyyk8guBSH1-u5GD7IFnDjX6OF_P27qFWGi0JItH8.aJZWyp6yLOf1GGe-GjjxzfJyID7U-uoSATY_v7L5niA&dib_tag=se&keywords=16+port+managed+switch&qid=1727128401&s=electronics&sprefix=16+port+managed+switch%2Celectronics%2C147&sr=1-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1
>
> https://www.amazon.com/D-Link-Ethernet-Internet-Mountable-DGS-1100-16V2/dp/B0876G7X18/ref=sr_1_13?crid=1FR75LFWP0AC5&dib=eyJ2IjoiMSJ9.s8A9GIIlTdzvlq-xrnzobe0Ej98aWbxKaIlQ_VBJLcneeTKeLYaYyrXMA7XA1U6TlTSttS2KqM3F-2f19qIpJY7PqSuieOzvMoAWO5kGsyxF3DYI5mrKvSvnzN3QrOwvPmXkJS7cIUBw-pzM0tNK0X-sKaAgdEX0AwvsZ9e0kdUTKFomC21eDkdjqTN4v2YXA6rRB9NYO20THzQ__5nhRlQAxT9CDR_4diZqlIHPOIaZ5E9zfJTzrWiE7ye7C2K_dOqyyk8guBSH1-u5GD7IFnDjX6OF_P27qFWGi0JItH8.aJZWyp6yLOf1GGe-GjjxzfJyID7U-uoSATY_v7L5niA&dib_tag=se&keywords=16%2Bport%2Bmanaged%2Bswitch&qid=1727128401&s=electronics&sprefix=16%2Bport%2Bmanaged%2Bswitch%2Celectronics%2C147&sr=1-13&th=1
>
> All in the ~$70-110 price range.
>
>
>

davidelang, 2024-09-23:
Owen DeLong wrote:

> Since a 5 port switch is effectively a 4 port (plus uplink) switch, the $/port is actually about the same as the 16 port switches I was proposing in the $70 range. $70/15+ usable = $4.67/port while [...]

I'm more interested in what the chipset inside the box is than the brand name on 
the outside :-)

but we'd have to buy one or hunt down a reference to find that out

the 8 port one for $16 is $2.29 per port PoE
https://www.amazon.com/STEAMEMO-Easy-Managed-Ethernet-Splitter/dp/B0DGKN4L2F?th=1

but as I said, I haven't tried one of these yet, so they could be utter junk.

David Lang