socallinuxexpo/scale-network #806: ZTP over the me0 port for switches

sarcasticadmin, 2024-12-06:
## Description

From the second work party there was discussion around ZTP

Prototype KEA config that will allow for loading of a zeroized ex4200-48p (or px) with the specified JunOS version and a copy of miniconfig through ZTP over the me0 port.

## Acceptance Criteria

- miniconfig is able to be loaded through ZTP over the me0 port

owendelong, 2024-12-12:
Kea Config has been prototyped and tested in my home environment. Kea stanzas still need to be isolated and optimized for handling more than one type of switch (for example, current setup handles only ex4200-48p, minor additional configuration is needed for ex4200-48px, and more significant configuration will be needed to add the ability to handle ex4300-48p).

I'm hoping that with some luck, I'll be able to get at least one ex4300-48p into a test environment soon.

owendelong, 2024-12-12:
Here are all the additions to kea-dhcp4.conf to make ZTP work.
`
{
  "Dhcp4": {
       ...
    // Vendor Specific and Custom Option Definitions (Global)
    "option-def": [
      // Juniper ZTP Options
      {   
        "name": "image-file-name",
        "code": 0,
        "space": "Juniper-ZTP",
        "type": "string",
        "record-types": "",
        "array": false
      },    
      {     
        "name": "config-file-name",
        "code": 1,
        "space": "Juniper-ZTP",
        "type": "string",
        "record-types": "",
        "array": false
      },  
      {     
        "name": "image-file-type",
        "code": 2,
        "space": "Juniper-ZTP",
        "type": "string",
        "record-types": "",
        "array": false
      },
      {
        "name": "transfer-mode",
        "code": 3,
        "space": "Juniper-ZTP",
        "type": "string",
        "record-types": "",
        "array": false
      },
      {
        "name": "alt-image-file-name",
        "code": 4,
        "space": "Juniper-ZTP",
        "type": "string",
        "record-types": "",
        "array": false
      },
      {
        "name": "http-port",
        "code": 5,
        "space": "Juniper-ZTP",
        "type": "string",
        "record-types": "",
        "array": false
      },
      {
        "name": "ftp-timeout",
        "code": 7,
        "space": "Juniper-ZTP",
        "type": "string",
        "record-types": "",
        "array": false
      }
    ],
    ...
    "subnet4": [
      ...
      {
        "id": 4,
        "subnet": "192.168.255.0/24",
        "option-data": [
          {
            "space": "dhcp4",
            "name": "subnet-mask",
            "code": 1,
            "data": "255.255.255.0"
          },
          {
            "space": "dhcp4",
            "name": "tftp-server-name",
            "code": 66,
            "data": "192.168.255.10"
          }
        ],
        "pools": [
          {
            "pool": "192.168.255.20 - 192.168.255.80"
          }
        ],
        "client-class":  "Juniper-ZTP-ex4200-48p"
      }
    ],
    "client-classes": [
      ...
      {
        "name": "Juniper-ZTP-ex4200-48p",
        "test": "option[60].text == 'Juniper-ex4200-48p'",
        "option-def": [
          {
            "name": "vendor-encapsulated-options",
            "code": 43,
            "type": "empty",
            "encapsulate": "Juniper-ZTP"
          }
        ],
        "option-data": [
          {
            "name": "config-file-name",
            "space": "Juniper-ZTP",
            "data": "config/miniconfig"
          },
          {
            "name": "image-file-name",
            "space": "Juniper-ZTP",
            "data": "images/jinstall-ex-4200-15.1R7.9-domestic-signed.tar.gz"
          },
          {
            "name": "transfer-mode",
            "space": "Juniper-ZTP",
            "data": "http",
          },
          {
            "name": "tftp-server-name",
            "data": "192.168.255.20"
          },
          {
            "name": "vendor-encapsulated-options"
          }
        ]
      }
    ]
  }
}
`

I also set up a simple apache instance to serve up config/miniconfig and the install image from images/jinstall-ex4200...

These were placed in the default /var/www/html HTML Root that apache2 packages for linux ship with.

IIRC, that was all that was needed in order to make ZTP work. My recommendation would be to add both of these capabilities to the stock image we use for NIX to contain the tools to flash switches.

owendelong, 2026-01-15:
Significant progress has been made on this and I hope to have a full ZTP solution in the coming weeks (hopefully before the work party).