socallinuxexpo/scale-network #812: deprecate magic nix cache in github actions

sarcasticadmin, 2025-01-23:
We received the following email:

```
Hi ...I'm reaching out about Magic Nix Cache
<https://github.com/DeterminateSystems/magic-nix-cache> (MNC) from
Determinate Systems. I believe at least one SCALE repo is leveraging the
tool, and I wanted to give you a heads up about a breaking change, but I
know you may not be the right person.

In short, GitHub has changed their Actions Cache API and when they
deprecate the old API after Jan 31st, Magic Nix Cache will no longer work
for some users. The good news is that many users are unaffected and for
those that are, there is a solution. More info here
<https://determinate.systems/posts/magic-nix-cache-free-tier-eol/>.

If there is anything we can help SCALE with so that this change doesn't
impact you, we'd love the chance to do that and hear your feedback on the
tool in general.

Thanks for your time and help getting this to the right person. See you at
SCALE in March!

...
Determinate Systems
```

We should figure out what to do with magic cache since its deprecated after Jan 31 2025. We are using it in the following place in ci: https://github.com/socallinuxexpo/scale-network/blob/6f65d7bc0e90f8fe341a5a449e2806c176a5dd4d/.github/workflows/ci.yml We might not need these actions any longer since we are using a gitlab hosted running for executing the entire test suite: https://github.com/socallinuxexpo/scale-network/blob/6f65d7bc0e90f8fe341a5a449e2806c176a5dd4d/.gitlab-ci.yml#L10-L17 we would need to enable these checks for PRs since I think its only running on changes to  `master` 

We could also setup a cache on the dev-server and serve it up to github actions although I think that might still be pretty slow since it wont be local to the github actions runners.

## AC

- Identify path forward after magic cache
- Implement necessary next steps

kylerisse, 2025-01-23:
> We might not need these actions any longer since we are using a gitlab hosted running for executing the entire test suite

From an HA standpoint we should probably keep the Github Actions for cases when `dev-server` might be down, especially during the show when concurrent PRs are flying in. Stranger things have happened. Ideally the solution would be a redundant physical server to distribute the runner loads on.

Short of that, perhaps we should take up DetSys on their offer for flakehub caching, if it is as painless as they claim. Their announcement indicates we can request a free tier for open source projects. Could be a win-win for both sides.

> We could also setup a cache on the dev-server and serve it up to github actions although I think that might still be pretty slow since it wont be local to the github actions runners.

Another solid option depending on the definition of "slow" here with the only downside being even more eggs in the same basket.

sarcasticadmin, 2025-01-24:
> From an HA standpoint we should probably keep the Github Actions for cases when dev-server might be down, especially during the show when concurrent PRs are flying in. Stranger things have happened. [...]

Good point, Id be open to exploring flakehub caching

sarcasticadmin, 2025-01-29:
Actual notice from github (4/16/2024): https://github.blog/changelog/2024-12-05-notice-of-upcoming-releases-and-breaking-changes-for-github-actions/

sarcasticadmin, 2025-06-18:
Maybe we dont have to after all: https://determinate.systems/posts/bringing-back-magic-nix-cache-action/

closed 2025-09-10 by sarcasticadmin