socallinuxexpo/scale-network #827: enable password prompt login for openwrt serial tty

sarcasticadmin, 2025-02-02:
Currently serial tty access does not prompt for a username and password.

We didnt have this issue since most of our devices didnt have an exposed serial port. However with the OpenWrt One it exposes a serial usb-c port right on the front.

https://openwrt.org/docs/guide-user/security/openwrt_security#securing_tty_and_serial_console

## AC
- Enforce serial tty login

davidelang, 2025-02-02:
in /etc/config/system this results in:

config system
        option hostname 'OpenWrt'
        option ttylogin '1'
        option log_size '128'
        option urandom_seed '0'
        option zonename 'UTC'
        option log_proto 'udp'
        option conloglevel '8'
        option cronloglevel '7'

sarcasticadmin, 2025-02-06:
Completed as #828

closed 2025-02-06 by sarcasticadmin