socallinuxexpo/scale-network #877: Consider using network-dispatcher to add mass flash interfaces to bridge

owendelong, 2025-03-05:
network-dispatcher is a dbus subscriber that monitors network events and executes scripts with environment variables set based on the event.
This would allow us to detect new interfaces and add them to the bridge as they are added to the device hosting mass flash.

https://gitlab.com/craftyguy/networkd-dispatcher

It's available as a package in most distros, including Debian. I don't know if there's a package for Nix.

owendelong, 2025-03-05:
I figure we look for "Carrier Detect" type messages and then wait 60 seconds, see if the interface has a valid (non-Link Local) address after that time, and if not, add it to the bridge.

Scenarios I've considered so far:
1. Laptop has interface we want to use for real networking -- Should get DHCP address within ~30 seconds after connection, set up networkd-dispatcher scripts to wait 60 seconds for a valid DHCP address before taking action to add new interface to bridge.
2. Laptop has interface we aren't using for mass flash that is normally used for real networking -- Adding this to the bridge should be mostly harmless. Worst case scenario I've been able to imagine so far is DHCP server ends up assigning an address to that interface and it starts trying to flash it as an AP. Since the SSH auth will certainly fail, this should be a non-issue.
3. An interface that is desired for real networking isn't connected for the first 60 seconds after loading mass flash. There is a possibility that connecting this to a live network could interfere with other clients on the live network by giving them bad addresses and trying to flash them. Documentation should include a caveat to shutdown the host prior to connecting a real network and then reboot the host after physical connection is completed.

MrHamel, 2025-03-05:
> Caution: This is an external email and may be malicious. Please take care when clicking links or opening attachments.

The #1 option is very similar to how eero Wi-Fi mesh routers work.

They have a single multi-gig and single 1g port on them like the OpenWRT One, and either of them can be WAN or LAN. I found that one out when I tried to set one up out of the box and couldn't DHCP an address on either port from the device and had to do device setup via their mobile app which connects to the access point, to proceed forward.